MarlinFirmware/Marlin #24357: X_AXIS_TWIST_COMPENSATION causes Warning to be generated when compiling Marlin 2.1

CBoismenu, 2022-06-16:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I have been working on compiling a version of Marlin 2.1 and was able to do so without getting any errors or warning.

The problem starts if i try to enable X_AXIX_TWIST_COMPENSATION.

When i try to compile the firmware I get the following warning:

Marlin\src\lcd\menu\menu_x_twist.cpp: In function 'void xatc_wizard_menu()':
Marlin\src\lcd\menu\menu_x_twist.cpp:104:19: warning: unused variable 'digs' [-Wunused-variable]
  104 |     const uint8_t digs = !UNEAR_ZERO((FINE_MANUAL_MOVE) * 1000 - int((FINE_MANUAL_MOVE) * 1000)) ? 4 :
         |                          ^~~~

Has anybody seen that and if so is it normal, if not would anyone have any idea on how to remedy the situation.

Thanks,

### Bug Timeline

new

### Expected behavior

Expected no warning to be generated when compi;ing the firmware

### Actual behavior

See generated warning message when compiling the firmware.

### Version of Marlin Firmware

2.1

### Printer model

Prusa bear clone with bltouch

### Electronics

SKR2

### Bed Leveling

ABL Bilinear mesh

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/8914380/Marlin.zip)

thisiskeithb, 2022-06-16:
The usage of `digs` was removed in https://github.com/MarlinFirmware/Marlin/pull/24278/, but the warning can be safely ignored.

CBoismenu, 2022-06-16:
I also modified the pins_BTT_SKR_V2_0_common.h file to add an I2C EEPROM to the board fili is attache
[pins_BTT_SKR_V2_0_common.zip](https://github.com/MarlinFirmware/Marlin/files/8914555/pins_BTT_SKR_V2_0_common.zip)
d

CBoismenu, 2022-06-16:
Thank you for the quick reply. Will try the compiled version on my printer and see how it goes.

closed 2022-06-16 by CBoismenu

thisiskeithb, 2022-06-16:
I've submitted https://github.com/MarlinFirmware/Marlin/pull/24358 to remove the unused `digs` variable.