MarlinFirmware/Marlin #24361: [BUG] (bug summary) Power Less Recovery restart the machine and stop the printer.

Fenainternet, 2022-06-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a simple Ender3 V4.2.2 [STM32F103RET6 (9U MXS 99 007) ARM2] [https://www.screenpresso.com/=TQ1Ve](url) [https://www.screenpresso.com/=sqjce](url)
On Visual Studio [https://www.screenpresso.com/=fNBR](url) I turn on the function power less recovery
When I start printing after 3 seconds the printing stops and the machine resets. 
I use the original sdcard from creality send it with the machine.

### Bug Timeline

this is new

### Expected behavior

The normal behavior should be after the interruption of the power supply and its restabilization, the printer would purge the filament and resume printing where it left off.

### Actual behavior

The printer has started the printing process normally. It starts by purging the initial filament on the left side of the printer normally. At this moment, when printing the line with the purge about 10 cm, the printer stops and the display is turned off and the machine restarts. After that, it is ready to start a new print. Repeating printing with other files already printed in the past process it happens again. When I turned off the Power Less Recovery on Visual Studio, every thing came back to normal.

### Steps to Reproduce

1. On visual studio turn on the Power Less Recovery [https://www.screenpresso.com/=ofRLf](url
2. Star the compilation
3. copy the bin file to sdcard
4. put the sdcard on ender 3
5. waiting for de stard
6. select a file
7. start the printing
8. after heating the bad and hotend the print starts
9. Start with the norlmal star purge
10. After  about 10 cm of purg in line on left of bad the printer stops
11. Restart it self 
12. Choose another file to print
13. Start the printing
14. the problem will occours again.

### Version of Marlin Firmware

Marlin 2.0.9.4

### Printer model

Ender 3

### Electronics

Ender3 V4.2.2 [STM32F103RET6 (9U MXS 99 007) ARM2] [https://www.screenpresso.com/=TQ1Ve](url) [https://www.screenpresso.com/=sqjce](url)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-06-17:
you have   "#define SDCARD_READONLY                   // Read-only SD card (to save over 2K of flash)"

You cannot have sd card set to read only when you have Power Less Recovery, as it writes it gcode position data to the sdcard for recovery.

When comping you would have had 100's of these warning go past in yellow
#warning "Either disable SDCARD_READONLY or disable POWER_LOSS_RECOVERY."
Why did you ignore them?

closed 2022-06-17 by ellensp

Fenainternet, 2022-06-18:
I did ignore them! I disble SDCARD_REANDONLY... them the warning in yellow stopped... but the printer continues stop the impression at the starts. Nothing change.