MarlinFirmware/Marlin #24364: M28 sd card problem

unknownblue, 2022-06-17:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I'm using the M28 to write some gcode to a sd card, I build a simple host to send the commands but if I send the first line of gcode that I want to write to the file works fine, but if I close with M29 and do other stuff like move some motors and then re-open with M28 and write some line again and close the file this ''second'' line actually doesn't write, I opened the sd card on my computer and only the first line was written, it's weird because I think the firmware knows the line was before closing. sorry if this is the wrong place for support

### Version of Marlin Firmware

2.0.9.3

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-06-19:
No config files provided

parallyze, 2022-06-19:
Sorry for adding noise here - but I do have a question about what the thread starter is trying to do:

- Create / write to a file, end using M29
- Do other things/commands
- Open the previous file and APPEND data to it

I have my printers rarely connected to anything except power (SD WiFi cards), so maybe I'm missing
something out here. But if using something like Simplify3D to upload to my SD card and simply uploading
the same file again - I end up with an identical copy, not a file having the same content twice. And as far
as I can see from the control panel it is using M28 -> gcode -> M29.

Looking at cardreader.cpp:
`if (file.open(diveDir, fname, O_CREAT | O_APPEND | O_WRITE | O_TRUNC)) {`

...and the descriptions inside SdBaseFile.cpp, I wonder if it's even supported to append
to an existing file...? O_TRUNC seems to be what's used, otherwise uploading a file from
the host two or three times would result in a single file containing everything two or
three times...? And I can't see a way to tell the printer to append otherwise, at least from
the M28 documentation.

Again - sorry if this is distracting, not trying to hijack this issue report... also I'm relatively
new to Marlin, so I might just have not seen other relevant parts inside the files yet ^^

unknownblue, 2022-06-19:
> No config files provided no idea on platform/motherboard used.

I'm using a ramps 1.4 arduino mega and the marlin is the 2.0.9.4, and I checked the bugfix 2.1x didn't worked too the configs of the sd card are normal just uncommented the sd support

unknownblue, 2022-06-19:
> Sorry for adding noise here - but I do have a question about what the thread starter is trying to do:
>
> * Create / write to a file, end using M29
> * Do other things/commands
> * Open the previous file [...]

thisiskeithb, 2022-06-19:
@unknownblue Zip up and attach your configs in your next reply.

unknownblue, 2022-06-19:
> @unknownblue Zip up and attach your configs in your next reply.
[config files.zip](https://github.com/MarlinFirmware/Marlin/files/8936116/config.files.zip)

closed 2022-08-30 by github-actions[bot]