MarlinFirmware/Marlin #24365: [BUG] Latest release is not latest

Temtaime, 2022-06-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Currently marlin v1 is marked at latest at github, while v2.1 should be. This poses problems at least for me  :) 

### Version of Marlin Firmware

not matter

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-06-17:
Marlin `1.0.2-3` was tagged & released after `2.1`, which is correct. If the [Releases](https://github.com/MarlinFirmware/Marlin/releases) page is confusing, use the [download page on Marlinfw.org](https://marlinfw.org/meta/download/) instead.

closed 2022-06-17 by thisiskeithb

Temtaime, 2022-06-17:
I was using https://api.github.com/repos/MarlinFirmware/Marlin/releases/latest to get sources from latest release, maybe can you provide some link on the site which points always to latest release ?

thisiskeithb, 2022-06-17:
> I was using https://api.github.com/repos/MarlinFirmware/Marlin/releases/latest to get sources from latest release

Marlin `1.0.2-3` was tagged & released after `2.1`, which is correct. Until the next version of `2.1.x` is tagged & released, `1.0.2-3` will be the latest release.

> maybe can you provide some link on the site which points always to latest release ? Currently there're none

Use the [download page on Marlinfw.org](https://marlinfw.org/meta/download/) instead.