MarlinFirmware/Marlin #24369: stm32f407vet6+25q16 +st7796 spi lvgl not work

minhngoc123, 2022-06-19:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I USE BOARD STM32F407 WITH ST7796 SPI ,25q16 flash
WHEN I USE TFT_COLOR_UI EVERYTHING WORKS GOOD
BUT WITH
TFT_LVGL_EVERYTHING IS WRONG
I HAVE COPPED SUCCESSFULLY. THE SCREEN IS SUCCESSFULLY TO FEED FILE.BIN IN THE ASSETS dir, BUT IT DOESN'T seem to work.
[image]
[image]
[image]

MY CON FILE FILE PIN :
/**

Marlin 3D Printer Firmware
Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
Based on Sprinter and grbl.
Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
This program is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.
This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
GNU General Public License for more details.
You should have received a copy of the GNU General Public License
along with this program. If not, see https://www.gnu.org/licenses/.
*/
#pragma once

/**

STM32F407VET6 with RAMPS-like shield
'Black' STM32F407VET6 board - https://www.stm32duino.com/viewtopic.php?t=485
Shield - https://github.com/jmz52/Hardware
*/
#define ALLOW_STM32DUINO
#include "env_validate.h"

#if HOTENDS > 2 || E_STEPPERS > 2
#error "Black STM32F4VET6 supports up to 2 hotends / E steppers."
#endif

#ifndef BOARD_INFO_NAME
#define BOARD_INFO_NAME "Black STM32F4VET6"
#endif

#define DEFAULT_MACHINE_NAME "STM32F407VET6"

//#define I2C_EEPROM
#define SRAM_EEPROM_EMULATION
#define MARLIN_EEPROM_SIZE 0x2000 // 8K

//
// Servos
//
//#define SERVO0_PIN PC6
#define SERVO1_PIN PC7

//
// Limit Switches
//
#define X_MIN_PIN PC13
#define X_MAX_PIN PA15
#define Y_MIN_PIN PA5
#define Y_MAX_PIN PD12
#define Z_MIN_PIN PD14
#define Z_MAX_PIN PD15

//
// Steppers
//
#define X_STEP_PIN PC4
#define X_DIR_PIN PA4
#define X_ENABLE_PIN PE7

#define Y_STEP_PIN PE5
#define Y_DIR_PIN PE2
#define Y_ENABLE_PIN PE6

#define Z_STEP_PIN PD5
#define Z_DIR_PIN PD3
#define Z_ENABLE_PIN PD6

#define E0_STEP_PIN PD7
#define E0_DIR_PIN PD0
#define E0_ENABLE_PIN PB9

#define E1_STEP_PIN PE0
#define E1_DIR_PIN PE1
#define E1_ENABLE_PIN PB8

//
// Temperature Sensors
//
#define TEMP_0_PIN PC0 // T0
#define TEMP_1_PIN PC1 // T1
#define TEMP_BED_PIN PC2 // TB

#ifndef TEMP_CHAMBER_PIN
#define TEMP_CHAMBER_PIN PC3 // TC
#endif

//
// Heaters / Fans
//
#define HEATER_0_PIN PA2 // Heater0
#define HEATER_1_PIN PA3 // Heater1
#define HEATER_BED_PIN PA1 // Hotbed

#define FAN_PIN PE9 // Fan0
#define FAN1_PIN PE11 // Fan1
#define FAN2_PIN PE13 // Fan2
#define FAN3_PIN PE14 // Fan3

//
// Misc. Functions
//
#define LED_PIN PA6
//#define LED_PIN PA7
#define KILL_PIN PB1

//
// LCD / Controller
//
//#define SD_DETECT_PIN PC5
//#define SD_DETECT_PIN PA8 // SDIO SD_DETECT_PIN, external SDIO card reader only

//#define BEEPER_PIN PD10
#define LCD_PINS_RS PE15
#define LCD_PINS_ENABLE PD8
#define LCD_PINS_D4 PE10
#define LCD_PINS_D5 PE12
#define LCD_PINS_D6 PD1
#define LCD_PINS_D7 PE8
#define BTN_ENC PD9
#define BTN_EN1 PD4
#define BTN_EN2 PD13

#define TFT_CS_PIN LCD_PINS_ENABLE
#define TFT_A0_PIN TFT_DC_PIN
#define TFT_SCK_PIN PB13 // SPI2_SCK
#define TFT_MISO_PIN PB14 // SPI2_MISO
#define TFT_MOSI_PIN PB15 // SPI2_MOSI
#define TFT_DC_PIN PC6
#define TFT_RST_PIN PD10
//
#define LCD_READ_ID 0xD3
#define LCD_USE_DMA_SPI

#define TFT_BUFFER_SIZE 14400

#define TOUCH_CS_PIN LCD_PINS_D5 // PE12
#define TOUCH_SCK_PIN PB13 // SPI2_SCK
#define TOUCH_MISO_PIN PB14 // SPI2_MISO
#define TOUCH_MOSI_PIN PB15 // SPI2_MOSI
// Onboard SD support
//
#define SPI_FLASH
#if ENABLED(SPI_FLASH)
#define HAS_SPI_FLASH 1
#define SPI_DEVICE 2
#define SPI_FLASH_SIZE 0x1000000
#define SPI_FLASH_CS_PIN PB0
#define SPI_FLASH_MOSI_PIN PB5
#define SPI_FLASH_MISO_PIN PB4
#define SPI_FLASH_SCK_PIN PB3
#endif
#ifndef SDCARD_CONNECTION
#define SDCARD_CONNECTION ONBOARD
#endif

#if SD_CONNECTION_IS(ONBOARD)
#define SDIO_SUPPORT
#define SDIO_D0_PIN PC8
#define SDIO_D1_PIN PC9
#define SDIO_D2_PIN PC10
#define SDIO_D3_PIN PC11
#define SDIO_CK_PIN PC12
#define SDIO_CMD_PIN PD2
#endif

### Bug Timeline

NEW

### Expected behavior

I HOPE IT FIX

### Actual behavior

[image]
[image]
[image]

### Steps to Reproduce

.

### Version of Marlin Firmware

2.0.9.3

### Printer model

PRUSA

### Electronics

BLACK_STM32F407VE

### Add-ons

.

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8934594/Configuration.zip)

thisiskeithb, 2022-06-19:
> Did you test the latest bugfix-2.1.x code?
> No, but I will test it now!

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

rhapsodyv, 2022-06-19:
Please try format the sd card (or try with another sd) and fully copy only the assets folder (with all files included), and test it.
The images you shared shows that the assets wasn’t loaded.
Or there’s an issue with the spi flash communication. You can try lower the spi speed for the spi flash to test.

minhngoc123, 2022-06-19:
i try bugfix-2.1.x but it same,i coppy the folder assets it work,but the picture is distorted, and after rebooting everything is wrong again
https://user-images.githubusercontent.com/70738064/174463131-abd340a5-300f-4241-b521-0baccaad9243.mp4
when i reboot
[image]

closed 2022-08-29 by github-actions[bot]