MarlinFirmware/Marlin #24372: M122 S0 doesn't stop or prevent continuous debugging output

CBoismenu, 2022-06-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With TMC_DEBUG enabled in Configuration_adv.h using the M122 S0 command doesn't prevent continuous debugging from being trigered

### Bug Timeline

Was also there in 2.0.9.3

### Expected behavior

Expecting printer not to enter Continuous Debugging Mode.

### Actual behavior

M122 S0 or M122 S1 produces the same output system enters Continuous Debugging Mode

### Steps to Reproduce

Enable TMC_DEBUG in Configuration_adv.h

Compile and load the firmware

M122 S0 in Octoprint Terminal causes the following to be generated:

Send: M122 S0
Recv: axis:pwm_scale/curr_scale/mech_load|flags|warncount
Recv: X Y Z Z2 E
Recv: Address 0 0
Recv: Enabled false false true true false
Recv: Set current 550 550 200 200 550
Recv: RMS current 520 520 183 183 520
Recv: MAX current 733 733 258 258 733
Recv: Run current 16/31 16/31 5/31 5/31 16/31
Recv: Hold current 8/31 8/31 2/31 2/31 8/31
Recv: CS actual 8/31 8/31 2/31 2/31 8/31
Recv: PWM scale
Recv: vsense 1=.18 1=.18 1=.18 1=.18 1=.18
Recv: stealthChop true true true true true
Recv: msteps 16 16 16 16 16
Recv: interp true true true true true
Recv: tstep max max max max max
Recv: PWM thresh. 98 98 98 98 188
Recv: [mm/s] 100 100 20 20 30
Recv: OT prewarn false false false false false
Recv: triggered
Recv: OTP false false false false false
Recv: pwm scale sum 12 11 11 10 10
Recv: pwm scale auto 0 0 0 1 0
Recv: pwm offset auto 45 42 127 106 36
Recv: pwm grad auto 13 12 25 23 14
Recv: off time 4 4 4 4 4
Recv: blank time 24 24 24 24 24
Recv: hysteresis
Recv: -end 2 2 2 2 2
Recv: -start 1 1 1 1 1
Recv: Stallguard thrs 60 60
Recv: uStep count 408 424 360 824 88
Recv: DRVSTATUS X Y Z Z2 E
Recv: sg_result 60 110
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: X 0xC0:08:00:00
Recv: Y 0xC0:08:00:00
Recv: Z 0xC0:02:00:00
Recv: Z2 0xC0:02:00:00
Recv: E 0xC0:08:00:00
Recv:
Recv:
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: Testing E connection... OK
Recv: ok
Recv: X:12/8/-|TI| Y:11/8/-|TI| Z:11/2/-|TI| Z2:10/2/-|TI| E:10/8/-|TI|
Recv: T:150.06 /150.00 B:60.00 /60.00 @:26 B@:0
Recv: Not SD printing
Recv: X:12/8/-|TI| Y:11/8/-|TI| Z:11/2/-|TI| Z2:10/2/-|TI| E:10/8/-|TI|
Recv: X:0.0000 Y:0.0000 Z:15.0000 E:0.0000 Count X:0 Y:0 Z:6023
Recv: X:12/8/-|GTI| Y:11/8/-|GTI| Z:11/2/-|GTI| Z2:10/2/-|GTI| E:10/8/-|GTI|
Recv: Not SD printing
Recv: X:12/8/-|TI| Y:11/8/-|TI| Z:11/2/-|TI| Z2:10/2/-|TI| E:10/8/-|TI|

Goes on continuously.

I can change the frequency with M122 P<> but canot stop it.

### Version of Marlin Firmware

2.1

### Printer model

Duplicator i3 plus

### Electronics

SKR V1.3

### Add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/8936217/Marlin.zip)

parallyze, 2022-06-19:
Can confirm this is happening in 2.1 release and bugfix from a few days ago. Setting the
interval using P does work, turning off reporting using S0 doesn't. Will test up to date
bugfix and provide configs if required.

Possibly connected to 25d5d2ab507075f282787c6e5c9d57f1a3972e1e

Changing the line
` const bool sflag = parser.seen_test('S'), sval = sflag && parser.value_bool();`
to 
` const bool sflag = parser.seen('S'), sval = sflag && parser.value_bool();`

("parser.seen_test -> parser.seen") seems to work, S0/S1 does turn off/on reporting again,
don't know what possible other effects this change might cause.

hcet14, 2022-06-21:
Latest bugfix-2.1.x (3 hours ago) same behavior.
[BUG] sg_result on y doesn't work.
I'll build a classical cartesian with 4 steppers. I changed the drivers, no difference.
Right now, I have the mainboard with the 4 steppers connected on the bench. If you want me to do some tests, tell me.

CBoismenu, 2022-06-21:
The proposed modification fixed the problem on my system.

I haven'seen any negative or unwanted effect at this point.

I'll keep you posted if I find anything wrong.

hcet14, 2022-06-22:
> I haven'seen any negative or unwanted effect at this point.

Did you try sg_result and hold the steppers?

hcet14, 2022-07-20:
@CBoismenu

Did you read my question? 

Did you try sg_result and hold the steppers?

CBoismenu, 2022-07-20:
@hcet14 

Did read your question.

Although I've been working on compiling the firmware for a while I didn't understand what you wanted me to try.

If you can give me more information I would be willing to try it out.

hcet14, 2022-08-07:
@CBoismenu

Sorry I was a little sick and then I had to work a lot, now ok.

Tried it with Marlin-bugfix-2.1.x_220807_17_52 from today:
> M122 S0 or M122 S1 produces the same output system enters Continuous Debugging Mode

As you stated at the beginning, nothing changed!

I guess nobody of the developers noticed the bug. But you didn't mark this issue as a bug!

> Although I've been working on compiling the firmware for a while I didn't understand what you wanted me to try. 
> If you can give me more information I would be willing to try it out. 

Please use M122 and look at these values, just an example (sg_result 206 16 58 86). Now, please hold/brake your y-axis and move it. Does the y-value changes? You have to use M122 again to read it out. Thanks in advance.

It's working on x, z, and e as expected. Mainboard is SKR 2 with TMC2209s and Cartesian printer. I swapped my TMC2209s around, but no difference. Y stays at 16! Maybe you may try y and tell me your results. SKR 2 has a lot of hardware bugs, maybe it's caused by the board.

Please edit your issue and put a big [BUG] in front of your bug summary! Good work!

CBoismenu, 2022-08-11:
Tried your request on two of my printers.

Configuration #1 SKR1.3 with TMC2209 on X and Y and TMC2208 on Z Z2 and E
Before Y move: sg_result 60 16
After Y move while holding the bed: sg-result 60 62

Configuration #2 SKR2.0 with TMC2209 on all axis
Before Y move: sg_result 0 142 0 0 0
After Y move while holding the bed: sg-result 0 42 0 0 0

Both boards are on Marlin 2.1

hcet14, 2022-08-12:
@CBoismenu
Thank you very much!
Mmmh, mysterious.
Which Marlin version did you use for your tests? bugfix-2.1.x from (date, time)? I would like to test the exact version you were using.

CBoismenu, 2022-08-12:
Marlin version was 2.1 for both boards

hcet14, 2022-08-15:
@CBoismenu

I'm terribly sorry that I stole your time! Thank you very much!

I thought my SKR 2 had a bug, but no. Just my missing knowledge about Marlin and TMCs.

I should have tried both directions from the beginning.

This is my setup 
[image]

I used Marlin-2.1.1 from 220806.

M122 results:
sg_result	0	0	0	0   after reboot, no stepper moved yet
sg_result	210	0	0	0   x to one direction
sg_result	16	0	0	0   x to the other direction
sg_result	16	212	0	0   y to one direction
sg_result	16	16	0	0   y to the other direction
sg_result	16	16	58	0   z to one direction
sg_result	16	16	70	0   z to the other direction
sg_result	16	16	70	88   e to one direction
sg_result	16	16	70	16   e to the other direction

Funny behaviour of z. But I don't care.

Thanks a lot. I read the TMC datasheet back and forth and didn't understand the behaviour.

I asked here ['https://reprap.org/forum/read.php?415,890279'](https://reprap.org/forum/read.php?415,890279) and here ['https://github.com/bigtreetech/SKR-2/issues/164'](https://github.com/bigtreetech/SKR-2/issues/164).

But with this version of Marlin, that I tried M122 S0 behaves like M122 S1. There is no continuous debugging. It just runs once.

hcet14, 2022-08-16:
Latest bugfix-2.1.x (5 hours ago) same behavior, as stated in first post. When you use M122 S1, it runs forever until reset of the mainboard.

closed 2022-10-27 by github-actions[bot]