MarlinFirmware/Marlin #24376: [BUG] Can't Compile for STM32 with DGUS Enabled

msticninja, 2022-06-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With #define DGUS_LCD_UI_MKS enabled, the compile fails.
Config: [Config2.1.x-6-20.zip](https://github.com/MarlinFirmware/Marlin/files/8943601/Config2.1.x-6-20.zip)

Output:
```
Marlin\src\lcd\extui\dgus\mks\DGUSScreenHandler.cpp:261:8: error: no declaration matches 'void DGUSScreenHandler::SDPrintingFinished()'
  261 |   void DGUSScreenHandler::SDPrintingFinished() {
      |        ^~~~~~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\mks\DGUSScreenHandler.cpp:261:8: note: no functions named 'void DGUSScreenHandler::SDPrintingFinished()'
In file included from Marlin\src\lcd\extui\dgus\mks\../mks/DGUSScreenHandler.h:24,
                 from Marlin\src\lcd\extui\dgus\mks\../DGUSScreenHandler.h:48,
                 from Marlin\src\lcd\extui\dgus\mks\DGUSScreenHandler.cpp:27:
Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSScreenHandlerBase.h:32:7: note: 'class DGUSScreenHandler' defined here
   32 | class DGUSScreenHandler {
      |       ^~~~~~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\mks\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandlerMKS::LanguageChange(DGUS_VP_Variable&, void*)':
Marlin\src\lcd\extui\dgus\mks\DGUSScreenHandler.cpp:419:73: error: call of overloaded 'WriteVariable(const uint16_t&, MKS_Choose)' is ambiguous
  419 |       dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE1, MKS_Language_Choose);
      |                                                                         ^
In file included from Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSScreenHandlerBase.h:24,
                 from Marlin\src\lcd\extui\dgus\mks\../mks/DGUSScreenHandler.h:24,
                 from Marlin\src\lcd\extui\dgus\mks\../DGUSScreenHandler.h:48,
                 from Marlin\src\lcd\extui\dgus\mks\DGUSScreenHandler.cpp:27:
Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSDisplay.h:66:15: note: candidate: 'static void DGUSDisplay::WriteVariable(uint16_t, int16_t)'
   66 |   static void WriteVariable(uint16_t adr, int16_t value);
      |               ^~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSDisplay.h:67:15: note: candidate: 'static void DGUSDisplay::WriteVariable(uint16_t, uint16_t)'
   67 |   static void WriteVariable(uint16_t adr, uint16_t value);
      |               ^~~~~~~~~~~~~

Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSDisplay.h:68:15: note: candidate: 'static void DGUSDisplay::WriteVariable(uint16_t, uint8_t)'
   68 |   static void WriteVariable(uint16_t adr, uint8_t value);
      |               ^~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSDisplay.h:69:15: note: candidate: 'static void DGUSDisplay::WriteVariable(uint16_t, int8_t)'
   69 |   static void WriteVariable(uint16_t adr, int8_t value);
      |               ^~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSDisplay.h:70:15: note: candidate: 'static void DGUSDisplay::WriteVariable(uint16_t, long int)'
   70 |   static void WriteVariable(uint16_t adr, long value);
      |               ^~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\mks\DGUSScreenHandler.cpp:420:75: error: call of overloaded 'WriteVariable(const uint16_t&, MKS_Choose)' is ambiguous
  420 |       dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE2, MKS_Language_NoChoose);
      |                                                                           ^
In file included from Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSScreenHandlerBase.h:24,
                 from Marlin\src\lcd\extui\dgus\mks\../mks/DGUSScreenHandler.h:24,
                 from Marlin\src\lcd\extui\dgus\mks\../DGUSScreenHandler.h:48,
                 from Marlin\src\lcd\extui\dgus\mks\DGUSScreenHandler.cpp:27:
Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSDisplay.h:66:15: note: candidate: 'static void DGUSDisplay::WriteVariable(uint16_t, int16_t)'
   66 |   static void WriteVariable(uint16_t adr, int16_t value);
      |               ^~~~~~~Compiling .pio\build\mks_robin_nano_v3\src\src\libs\crc16.cpp.o
~~~~~~
Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSDisplay.h:67:15: note: candidate: 'static void DGUSDisplay::WriteVariable(uint16_t, uint16_t)'
   67 |   static void WriteVariable(uint16_t adr, uint16_t value);
      |               ^~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSDisplay.h:68:15: note: candidate: 'static void DGUSDisplay::WriteVariable(uint16_t, uint8_t)'
   68 |   static void WriteVariable(uint16_t adr, uint8_t value);
      |               ^~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSDisplay.h:69:15: note: candidate: 'static void DGUSDisplay::WriteVariable(uint16_t, int8_t)'
   69 |   static void WriteVariable(uint16_t adr, int8_t value);
      |               ^~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSDisplay.h:70:15: note: candidate: 'static void DGUSDisplay::WriteVariable(uint16_t, long int)'
   70 |   static void WriteVariable(uint16_t adr, long value);
      |               ^~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\mks\DGUSScreenHandler.cpp:426:75: error: call of overloaded 'WriteVariable(const uint16_t&, MKS_Choose)' is ambiguous
  426 |       dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE1, MKS_Language_NoChoose);
[... 134 lines cut ...]
      |               ^~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSDisplay.h:69:15: note: candidate: 'static void DGUSDisplay::WriteVariable(uint16_t, int8_t)'
   69 |   static void WriteVariable(uint16_t adr, int8_t value);
      |               ^~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\mks\../mks/../DGUSDisplay.h:70:15: note: candidate: 'static void DGUSDisplay::WriteVariable(uint16_t, long int)'
   70 |   static void WriteVariable(uint16_t adr, long value);
      |               ^~~~~~~~~~~~~
*** [.pio\build\mks_robin_nano_v3\src\src\lcd\extui\dgus\mks\DGUSScreenHandler.cpp.o] Error 1
```

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

FLSun Super Racer

### Electronics

Stock Robin Nano V3

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-06-21:
A few issues here...

This compiles and seems right...  but I cannot test

Please test this

```DIFF
diff --git a/Marlin/src/lcd/extui/dgus/mks/DGUSScreenHandler.cpp b/Marlin/src/lcd/extui/dgus/mks/DGUSScreenHandler.cpp
index 18ac433b71..a44076b845 100644
--- a/Marlin/src/lcd/extui/dgus/mks/DGUSScreenHandler.cpp
+++ b/Marlin/src/lcd/extui/dgus/mks/DGUSScreenHandler.cpp
@@ -258,7 +258,7 @@ void DGUSScreenHandlerMKS::DGUSLCD_SendTMCStepValue(DGUS_VP_Variable &var) {
     ) filelist.refresh();
   }
 
-  void DGUSScreenHandler::SDPrintingFinished() {
+  void DGUSScreenHandlerMKS::SDPrintingFinished() {
     if (DGUSAutoTurnOff) {
       queue.exhaust();
       gcode.process_subcommands_now(F("M81"));
@@ -416,15 +416,15 @@ void DGUSScreenHandlerMKS::LanguageChange(DGUS_VP_Variable &var, void *val_ptr)
     case MKS_SimpleChinese:
       DGUS_LanguageDisplay(MKS_SimpleChinese);
       mks_language_index = MKS_SimpleChinese;
-      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE1, MKS_Language_Choose);
-      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE2, MKS_Language_NoChoose);
+      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE1, (uint8_t) MKS_Language_Choose);
+      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE2, (uint8_t) MKS_Language_NoChoose);
       settings.save();
       break;
     case MKS_English:
       DGUS_LanguageDisplay(MKS_English);
       mks_language_index = MKS_English;
-      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE1, MKS_Language_NoChoose);
-      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE2, MKS_Language_Choose);
+      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE1, (uint8_t) MKS_Language_NoChoose);
+      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE2, (uint8_t) MKS_Language_Choose);
       settings.save();
       break;
     default: break;
@@ -1446,12 +1446,12 @@ bool DGUSScreenHandlerMKS::loop() {
 void DGUSScreenHandlerMKS::LanguagePInit() {
   switch (mks_language_index) {
     case MKS_SimpleChinese:
-      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE1, MKS_Language_Choose);
-      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE2, MKS_Language_NoChoose);
+      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE1, (uint8_t) MKS_Language_Choose);
+      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE2, (uint8_t) MKS_Language_NoChoose);
       break;
     case MKS_English:
-      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE1, MKS_Language_NoChoose);
-      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE2, MKS_Language_Choose);
+      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE1, (uint8_t) MKS_Language_NoChoose);
+      dgusdisplay.WriteVariable(VP_LANGUAGE_CHANGE2, (uint8_t) MKS_Language_Choose);
       break;
     default:
       break;
diff --git a/Marlin/src/lcd/marlinui.h b/Marlin/src/lcd/marlinui.h
index 66e5e0c261..78ffbad429 100644
--- a/Marlin/src/lcd/marlinui.h
+++ b/Marlin/src/lcd/marlinui.h
@@ -502,11 +502,12 @@ public:
   #else // No LCD
 
     static void update() {}
-    static void return_to_status() {}
     static void kill_screen(FSTR_P const, FSTR_P const) {}
 
   #endif
 
+  static void return_to_status() {}
+
   #if ENABLED(SDSUPPORT)
     #if BOTH(SCROLL_LONG_FILENAMES, HAS_MARLINUI_MENU)
       #define MARLINUI_SCROLL_NAME 1
```

msticninja, 2022-06-21:
Thanks for the quick work. Embarrassingly, I cannot test either, since the screen seems to have bricked when updating it's firmware. I'll probably be ordering a new one, but I might just be done with it and use my old TFT35. This firmware brick is unrelated to Marlin, obviously, but I'll post the details for reference. 

It's an MKS H43, running a DWIN T5L chip. The screen went white while on the blue upgrade screen, while installing the latest MKS firmware. Can't get it to respond to any firmware files anymore, and it just boots to a white screen. I don't really have time to figure out if I can tap straight into the T5L and reprogram with DWIN tools, but if someone has a known method, let me know please.

If I can get a new H43 or unbrick the original, I'll try this diff as soon as possible.

ellensp, 2022-06-21:
Ive created a updated PR instead, diff removed.

ellensp, 2022-06-21:
closing this as pr has been created.

closed 2022-06-21 by ellensp