MarlinFirmware/Marlin #24392: #define PSU_CONTROL on BTT GTR V1.0: drives won't move, printer halts w/critical error

PGrave, 2022-06-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm currently on the final stage of building a BLV MGN Cube, with BTT GTR V1.0. After testing all calibrations, I went for power supply control feature. After enabling #define PSU_CONTROL and program  my machine steppers won't move any more. I hear a noise from drivers right after firmware programming ends and also if i order any movement afterwards, but no movement and printer halted ensues. I haven't noticed any other malfunctions with this firmware. After programming firmware without PSU Control enabled all drives work as expected.

### Bug Timeline

started with enabling PSU control on Marlin bugfix 2.1

### Expected behavior

I expected PSU control with a pin on board and GCode command to control.

### Actual behavior

Steppers will not move and printer halts with critical error after using firmware with PSU Control enabled. 

### Steps to Reproduce

1- Enable #define PSU_CONTROL;
2- compile for BTT GTR V1.0;
3- program firmware on board;
4- order any movement of printer;
5- ear grinding noise on steppers;
6- Printer halts with critical error;

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

BLV Mgn Cube with GTR V1.0

### Electronics

GTR V1.0; TFT50

### Bed Leveling

ABL Linear grid

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

PGrave, 2022-06-23:
[MarlinBfix2.1_PG.zip](https://github.com/MarlinFirmware/Marlin/files/8971786/MarlinBfix2.1_PG.zip)

zeleps, 2022-06-29:
Could you also attach your pins_BTT_GTR_V1_0.h file? I see you have dual Z motor but how is it connected? Also, I see no trinamic configuration for your 2209s, but they are not configured as standalone.

I have a GTR with PSU control and pretty loaded (2 z motors, 2 extruders, MAX31865, bltouch, and more) but I never had an issue. This is probably a misconfiguration issue. Attach the pins file, and I might be able to help.

PGrave, 2022-06-30:
Hello and thank you for you help! Here is the pins file you asked. I hope you can spot anything I might have spoiled! It is very possible, but between working and not working drives I only change that single define. I will try and do some more debugging on the machine, to get more or better info on the issue.
[pins_BTT_GTR_V1_0.zip](https://github.com/MarlinFirmware/Marlin/files/9021650/pins_BTT_GTR_V1_0.zip)

zeleps, 2022-07-20:
Hi, sorry for the delay, I was away from home for a while. Will look into it this weekend.

PGrave, 2022-07-23:
Hello and thank you @zeleps ! Appreciate any help you can provide. Have a nice weekend!

zeleps, 2022-07-23:
You have defined Z2_DRIVER_TYPE, but there is no definition for separate Z2 pins. I suspect you are running your dual Z motors on the same driver:
[image]
So, if you've plugged your 2nd Z motor on A, this is not a true dual motor drive, therefore you're only using one Z driver, and you should disable all Z2 config values in both configuration and configuration_adv.h.
If you want to have a true dual Z drive, you need to rename one of your Extruder drivers to Z2 (e.g. E2, B in the diagram above), like this (pins_btt_gtr1_0.h, @170):
```
#define Z2_STEP_PIN                         PD1
#define Z2_DIR_PIN                          PD0
#define Z2_ENABLE_PIN                       PD2
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                         PC12
#endif
```
This way you will have two independent Z motors, capable of moving independently (allowing you to enable G34 calibration).
This is an obvious misconfiguration, try fixing this and see if it solves your issue.

thisiskeithb, 2022-07-23:
@zeleps: If your board doesn’t have a native Z2, Marlin will map unused extruder pins to as Z2/3/etc., so there’s no need to make this pins change.

thisiskeithb, 2022-07-23:
Once powered up, send `M122 I` to reinitialize your drivers.

---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-07-23 by thisiskeithb

PGrave, 2022-07-26:
This might be user error, but I'm almost sure it's not. The issue is not only related to Z steppers, but all drives of the machine.
The thing is, if I don't enable #define PSU_CONTROL, machine works flawlessly, hours and hours of parts already done.
As soon as #define PSU_CONTROL is enabled, all drivers make a noise on boot and any movement order gets a "Printer Halted - Please Reset" error. No movement whatsoever occurs, only a grinding noise from driver.
Everything else performs correctly on printer, including PSU pin change state with command, just drivers will not move.