MarlinFirmware/Marlin #24394: Adjust for thermistor error

lutusp, 2022-06-24:
### Is your feature request related to a problem? Please describe.

Many printer parameters/values can be adjusted to accommodate a printer's mechanical properties -- one example is the rate of extrusion (M92 Ennn), to make extrusion behavior uniform and predictable.

But there's no similar adjustment for thermistor variations, and as it turns out, thermistors vary quit a bit, unit to unit. I have a printer  whose thermistor is typically 25 degrees in error, and there's no way to accommodate this error (and this is not at all untypical). As a result, I have to refer to a paper correction table when I want to specify a particular hot end temperature. It's all very 19th century.

Please consider adding a correction variable, not unlike the extrusion variable and others, so G-code can be written without constantly, manually converting between fantasy and reality.

This feature could have one of two forms:

* The simplest would be a single value -- an error offset -- that represents the average temperature error in a domain of interest.
* Better would be a two-element linear regression, i.e. the classic y = mx+b, where 'b' is the offset and 'm' is the slope.

Either one of these would be a very welcome upgrade to the G-code system, correcting what to me is a glaring omission.

Thanks for reading!

### Are you looking for hardware support?

No hardware changes.

### Describe the feature you want

See above.

thisiskeithb, 2022-06-24:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/13131

closed 2022-06-24 by thisiskeithb

lutusp, 2022-06-24:
@[thisiskeithb](https://github.com/thisiskeithb) thisiskeithb marked this as a duplicate of [#13131](https://github.com/MarlinFirmware/Marlin/issues/13131) [1 hour ago](https://github.com/MarlinFirmware/Marlin/issues/24394#event-6876907512)

So it seems this is already solved with M305 ...

Send: M305
Recv: echo:Unknown command: "M305"
Recv: ok

Or not (or only in the newest Marlin versions -- I have 20211227 and cannot upgrade).