MarlinFirmware/Marlin #24395: "the terminal terminated with exit code 1"  platform io error, while trying to compile marlin.

benjaminthefirst, 2022-06-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hey everyone. 
I was trying to flash my anet et4 pro on marlin. After opening the marlin folder on visual studio I tried to get into the platformio.ini file and change some codes which should have been presented somewhere about line 1000 (as i saw it on a tutorial). Unfortunately I found out that the file was only containing 286 lines of code which made me completely confused. There was no errors or problems at all and everything was looking fine. I tried to ignore the problem and clicked on build marlin but this error showed up: 

 Executing task: C:\Users\beny\.platformio\penv\Scripts\platformio.exe run 

Error: Invalid 'C:\Users\beny\Downloads\Marlin-bugfix-2.0.x\platformio.ini' (project configuration file): 'While reading from 'C:\\Users\\beny\\Downloads\\Marlin-bugfix-2.0.x\\platformio.ini' [line 284]: option 'build_flags' in section 'env' already exists'
The terminal process "C:\Users\beny\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

I tried to update my windows power shell and installed windows terminal app but those didn't help. I'm not familiar with coding and I have no idea how to fix this problem. I would greatly appreciate any help because I've been struggling with this problem for about a week.

by the way these are codes from line 281 to line 286 (last line) in platformio.ini file: 

(281) env:include_tree
(282) platform    = atmelavr
(283) board       = megaatmega2560
(284) build_flags = -c -H -std=gnu++11 -Wall -Os -D__MARLIN_FIRMWARE__
(285) src_filter  = +<src/MarlinCore.cpp>
(286) empty line

### Version of Marlin Firmware

Marlin-bugfix-2.0.x

### Printer model

anet et4 pro

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin-bugfix-2.0.x(1).zip](https://github.com/MarlinFirmware/Marlin/files/8983581/Marlin-bugfix-2.0.x.1.zip)

ellensp, 2022-06-25:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-06-25 by ellensp

Bob-the-Kuhn, 2022-06-25:
You're running from an outdated tutorial.  

After updating the names on some of the leveling items, your config builds without errors.  Attached are the updated files.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/8983652/Marlin.zip)

This is using the Marlin Auto Build plugin within VSCode.  There are some YouTube tutorials on this.