MarlinFirmware/Marlin #24396: Selecting Move Z now Selects/Moves X Instead in TFT_COLOR_UI

thisiskeithb, 2022-06-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Selecting Move Z now Selects/Moves X Instead in `TFT_COLOR_UI`

### Bug Timeline

https://github.com/MarlinFirmware/Marlin/commit/975c8f4b77ac91678ed102c75412a8868fc1689c

### Expected behavior

Selecting Move Z will move Z, not X.

### Actual behavior

Selecting Move Z will move/select X instead of Z.

### Steps to Reproduce

Test with any `TFT_COLOR_UI` config ([Biqu B1 SE Plus](https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/BIQU/B1%20SE%20Plus) or [Biqu BX](https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/BIQU/BX) as examples) and and try to move Z through the LCD.

### Version of Marlin Firmware

bugfix-2.1.x (0aebcc4)

### Printer model

Any TFT_COLOR_UI config

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Use [Biqu B1 SE Plus](https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/BIQU/B1%20SE%20Plus) or [Biqu BX](https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/BIQU/BX) as examples.

rhapsodyv, 2022-06-25:
I didn’t see that code, but I suspect that is similar to another menu issues I fixed in the past: the itemIndex variable being used for two different purpose.

thinkyhead, 2022-06-25:
I also suspect it is this. Looking for the culprit and a way to catch this kind of lost value in future.

rhapsodyv, 2022-06-25:
The issue, if I remember right, is triggered only when a redraw is needed.
The redraw call will re render the top menu, overwriting the itemIndex variable.
If there’s no redraw, the call keep inside the context that is using itemIndex for distance.

To fix it, I think a new way to hold the second state is needed. For example a parameter for the nested calls that is not dependent of itemIdex.

menus for multi index and mixing extruders have this same issue, already fixed in the past.

Farva42, 2022-06-25:
This issue also shows up with #define DWIN_MARLINUI_PORTRAIT same symptom and you can see the Axis label flip from Z to X as soon as you scroll down to a distance option.

thisiskeithb, 2022-06-25:
> This issue also shows up with #define DWIN_MARLINUI_PORTRAIT same symptom and you can see the Axis label flip from Z to X as soon as you scroll down to a distance option.

I'm going to guess that this bug affects anything not using the standard 12864/full graphic or 2004/character UIs.

thisiskeithb, 2022-06-26:
Closing since there’s now a PR to fix this: https://github.com/MarlinFirmware/Marlin/pull/24404

PR #24404 (2022-06-26): 🐛 Fix Manual Move axis selection
Fix a regression from 975c8f4b by explicitly sending axis indexes from the move menu items instead of relying on `itemIndex` persisting.

Fixes #24396
closed 2022-06-26 by thisiskeithb