MarlinFirmware/Marlin #24403: [FR] Is there any plan to integrate CR-FDM-v2.4.S1.200 board on Creality Sermoon V1?

duchengyao, 2022-06-26:
### Describe the feature you want

 Is there any plan to integrate CR-FDM-v2.4.S1.200 board on Creality Sermoon V1?

ellensp, 2022-06-26:
It already part of https://github.com/MarlinFirmware/Configurations/issues/649#issuecomment-1152876729 as you are well aware

duchengyao, 2022-06-26:
So there is no way to support my sermoon v1, until I get the source code.

ellensp, 2022-06-26:
You need a pins files that list what each and every pin does.

You can make this yourself, if your a electronics hobbyist or better. You literally track and trace each pin to the CPU to where it goes and document it 

You can also sometimes do this by reverse engineering the firmrware.bin 

Both of these options are last a resort and requires a lot of work

You can also pay someone to do this.Most will require payment in the form of a complete working machine.

duchengyao, 2022-07-02:
I have created the pin file for SermoonV1.

see https://github.com/Klipper3d/klipper/pull/5621

InsanityAutomation, 2022-07-02:
Pin file is just the standard 422 board. Nothing different there.

duchengyao, 2022-07-03:
> Pin file is just the standard 422 board. Nothing different there.

What `standard 422 board` means? Could you share link of the file?

ellensp, 2022-07-03:
@duchengyao 
ie it the same as pin out as BOARD_CREALITY_V4 in marlin
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/pins/stm32f1/pins_CREALITY_V4.h

ellensp, 2022-07-03:
we should probably add a motherboard alias for this....

duchengyao, 2022-07-03:
Really? I think it's definitely different
For example, In sermoon v1, [x_step_pin is PA7](https://github.com/Klipper3d/klipper/pull/5621/files#diff-f14c6fafe05df44cbe9d93498a9ae3fd5059f73e16572067cd71e7429b390c7fR20); In Ender3 and CREALITY_V4 is [PC2](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/pins/stm32f1/pins_CREALITY_V4.h#L101)

ellensp, 2022-07-03:
I agree these are nothing alike,  @InsanityAutomation  what are you talking about?

InsanityAutomation, 2022-07-03:
I've got a sermoon D1 and the board in it works with the creality 4.2.2 definition. Board is at the shop, and holiday weekend here in the US so its not in front of me, but I've got quite a few users running the precompiled builds on my GitHub for it. Can check my Creality Dwin2.0 branch and see it just sets the standard board.

duchengyao, 2022-07-03:
My config file is sermoon `V1` not `D1`,  a very confusing product number.

see https://3dprintbeginner.com/creality-sermoon-v1-pro-review/

InsanityAutomation, 2022-07-03:
Ah I missed the V in there. Different then!

duchengyao, 2022-07-04:
Am I allowed to create config file and submit PR?

InsanityAutomation, 2022-07-05:
Yup, anyone can make a PR!

duchengyao, 2022-09-29:
official source code has been released: https://github.com/CrealityOfficial/Sermoon-V1