MarlinFirmware/Marlin #24405: bug marlin 2.1.x Board : MKS TINYBEE Printer : Kossel Linear Plus Layer Shift Driver : TMC 2209 x4 (use a4988 in config)

flishy2532, 2022-06-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

print at 75% cura layershift  around layer line 70-80

### Bug Timeline

print at 75% cura layershift  around layer line 70-80

### Expected behavior

print at 75% cura layershift  around layer line 70-80

### Actual behavior

print at 75% cura layershift  around layer line 70-80

### Steps to Reproduce

print at 75% cura layershift  around layer line 70-80

### Version of Marlin Firmware

bugfix 2.1.x

### Printer model

Kossel Linear Plus

### Electronics

MKS TINYBEE TMC2209 x4

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

print at 75% cura layershift  around layer line 70-80

MKS TINYBEE Kossel Linear Plus TMC2209 x4 Marlin 2.1.x fixbug Layer Shift 1

https://youtu.be/foY1B9vRUCw

flishy2532, 2022-06-26:
[image]
[image]
[image]
[image]

flishy2532, 2022-06-26:
anyone can help fix ??
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8986396/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/8986397/Configuration_adv.zip)
?

ellensp, 2022-06-26:
If you have tmc2209's you use TMC2209_STANDALONE in marlin  not a4988 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-06-26 by ellensp