MarlinFirmware/Marlin #24408: can't receive "ok" from  prusa mmu2s in print ，in idle time is normal

fBn0523, 2022-06-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

我在串口助手发送T0可以收到ok的返回，但是使用sd卡进行打印相同的代码却陷入一个循环（加载，报错，重复加载）。我尝试了禁用耗材丝传感器，延长超时的时间，降低波特率。这都不起作用。
我分别对sd卡和串口信息进行模拟，最终的区别只体现在命令是由sd卡读取还是pc端发送，这令我非常不解。
难道是mairlin关于打印开始标志判断出现了错误，错过了信息返回？
具体代码在下面
M107；FAN OFF
G90 ; use absolute coordinates
M83 ; extruder relative mode
M140 S0 ; set final bed temp
M104 S150 ; set temporary nozzle temp to prevent oozing during homing
G4 S10 ; allow partial nozzle warmup
G28 ; home all axis
G1 Z50 F240
G1 X2 Y10 F3000
M104 S210 ; set final nozzle temp
M190 S0 ; wait for bed temp to stabilize
M109 S210 ; wait for nozzle temp to stabilize
G1 Z0.28 F240
G92 E0
G1 Y140 E10 F1500 ; prime the nozzle
G1 X2.3 F5000
G92 E0
G1 Y10 E10 F1200 ; prime the nozzle
G92 E0
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
T0//！！！！！！！！！！！！！！！！！！！！！！！这里就是错误发生的地方，我想他与上面的代码没有任何关联。
G1 Z.2 F9000
M107

### Bug Timeline

我认为这是一个新问题

### Expected behavior

23:11:36.964 -> MMU <= T0
23:11:38.958 -> echo:busy: paused for user
23:11:40.959 -> echo:busy: paused for user
23:11:42.959 -> echo:busy: paused for user
23:11:44.960 -> echo:busy: paused for user
23:11:46.961 -> echo:busy: paused for user
23:11:48.962 -> echo:busy: paused for user
23:11:50.963 -> echo:busy: paused for user
23:11:52.963 -> echo:busy: paused for user
23:11:54.964 -> echo:busy: paused for user
23:11:56.965 -> echo:busy: paused for user
23:11:57.745 -> MMU => 'ok'
23:11:57.745 -> MMU <= 'C0'
23:11:58.966 -> echo:busy: paused for user
23:12:00.865 -> MMU => 'ok'

### Actual behavior

23:16:59.912 -> MMU <= T0
23:16:59.912 -> echo:busy: paused for user
23:17:01.934 -> echo:busy: paused for user
23:17:03.924 -> echo:busy: paused for user
23:17:05.914 -> echo:busy: paused for user
23:17:07.903 -> echo:busy: paused for user
23:17:09.928 -> echo:busy: paused for user
23:17:11.918 -> echo:busy: paused for user
23:17:13.907 -> echo:busy: paused for user
23:17:15.932 -> echo:busy: paused for user
23:17:17.923 -> echo:busy: paused for user
23:17:19.912 -> echo:busy: paused for user
23:17:21.902 -> echo:busy: paused for user
23:17:23.927 -> echo:busy: paused for user
23:17:25.932 -> echo:busy: paused for user
23:17:27.919 -> echo:busy: paused for user
23:17:29.909 -> echo:busy: paused for user
23:17:31.933 -> echo:busy: paused for user
23:17:33.923 -> echo:busy: paused for user
23:17:35.913 -> echo:busy: paused for user
23:17:37.902 -> echo:busy: paused for user
23:17:39.927 -> echo:busy: paused for user
23:17:41.916 -> echo:busy: paused for user
23:17:43.933 -> echo:busy: paused for user
23:17:44.910 -> MMU retry
23:17:44.910 -> MMU not responding
23:17:44.910 -> MMU <= T0
23:17:45.923 -> echo:busy: paused for user

### Steps to Reproduce

发送T0 通过不同的方法 （SD卡，PC）来重现错误

### Version of Marlin Firmware

02010000

### Printer model

prusa mk 3s

### Electronics

mks tiny bee board

### Add-ons

prusa mmu2s

### Your Slicer

Slic3r

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/8986934/conf.zip)

GMagician, 2022-06-26:
Really weird since timeout should be 45s and your seems 5s can you confirm that `MMU_CMD_TIMEOUT` in mmu2.cpp is 45000L?
I don't have a MMU2 so I can't test it myself

GMagician, 2022-06-26:
Sorry for previous message. It seems you didn't get mmu2 answer after 45s. Sure You have no noise coming from steppers that generate EMI on serial?

fBn0523, 2022-06-27:
load the gcode above , cupy in sd and print in serial. find out the difference in printer motion
This file is output by the logic analyzer ->
[2022_6_27_data.zip](https://github.com/MarlinFirmware/Marlin/files/8993349/2022_6_27_data.zip)
the true  sign is that ->
CH0 muu_board -----data----> marlin board
CH1 marlin_board----data--->  muu board
[image]
but there is a signal missing
[image]

fBn0523, 2022-06-27:
@[GMagician](https://github.com/GMagician) 
It looks like Marlin missed some signal

GMagician, 2022-06-28:
Have you checked mmu response is correct? Marlin can timeout even if answer is not properly "ok". You may hack 'rx_str' function and echo what is really received (if any)

closed 2022-06-28 by fBn0523