MarlinFirmware/Marlin #24410: [BUG] During print LCD chirps and acts as if I pressed in the selection button

mrbass21, 2022-06-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm not sure what version started doing it, but my LCD is chirping during a print from the SD card (never tried direct) as though I had pressed in the button to browse the menu. It seems to happen randomly.

I'm using the stock config files with no changes and no hardware modifications or addons (yet).

### Bug Timeline

Unsure since I went from shipped firmware (2.0.8.2) to the most recent 2.1

### Expected behavior

Without interacting with the controls on the LCD during a print, there should be no phantom interaction.

### Actual behavior

The device seems to be getting phantom interaction.

### Steps to Reproduce

1. Produce a gcode on the SD card
2. Enter the SD card into the printer and start printing
3. At random intervals, I can hear the piezo chirp and the menu pops up on the LCD

Configuration files used (Should be stock from the bugfix-2.1.x branch):
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8987384/Configuration.zip)

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender 3 Pro

### Electronics

Everything is stock

### Add-ons

None

### Your Slicer

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

If anyone also has clarity on what motor drivers I have, the sticker on my SD card is "H8" which was confusing to me.

thisiskeithb, 2022-06-26:
I’d suggest routing your LCD cable away from any power sources as well as shielding the it or picking up an EMI filter board to reduce noise in the LCD cable.

closed 2022-06-26 by thisiskeithb

mrbass21, 2022-06-26:
Thanks. Thought it might have ben EMI too. I'll give that a shot.