MarlinFirmware/Marlin #24412: Bug MKS TINYBEE Layer Shift TMC2209 x4 StandAlone Kossel Linear Plus

flishy2532, 2022-06-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

MKS TINYBEE Kossel Linear Plus TMC2209 x4 Marlin 2.1.x fixbug Layer Shift 

LayerShift 75% Print On Cura Nozzle head is not go to same position 

Printer : Anycubic Kossel Linear Plus (Delta)
MotherBoard : MKS TINYBEE (ESP32 WIFI Support)
Driver : TMC2209 x4 (use TMC2209 standalone in config)
LCD : MKS mini v3.0 (RBG)

https://youtu.be/odYO6YuWUuE

### Bug Timeline

new bug

### Expected behavior

MKS TINYBEE Kossel Linear Plus TMC2209 x4 Marlin 2.1.x fixbug Layer Shift 

LayerShift 75% Print On Cura Nozzle head is not go to same position 

Printer : Anycubic Kossel Linear Plus (Delta)
MotherBoard : MKS TINYBEE (ESP32 WIFI Support)
Driver : TMC2209 x4 (use TMC2209 standalone in config)
LCD : MKS mini v3.0 (RBG)

https://youtu.be/odYO6YuWUuE

### Actual behavior

MKS TINYBEE Kossel Linear Plus TMC2209 x4 Marlin 2.1.x fixbug Layer Shift 

LayerShift 75% Print On Cura Nozzle head is not go to same position 

Printer : Anycubic Kossel Linear Plus (Delta)
MotherBoard : MKS TINYBEE (ESP32 WIFI Support)
Driver : TMC2209 x4 (use TMC2209 standalone in config)
LCD : MKS mini v3.0 (RBG)

https://youtu.be/odYO6YuWUuE

### Steps to Reproduce

MKS TINYBEE Kossel Linear Plus TMC2209 x4 Marlin 2.1.x fixbug Layer Shift 

LayerShift 75% Print On Cura Nozzle head is not go to same position 

Printer : Anycubic Kossel Linear Plus (Delta)
MotherBoard : MKS TINYBEE (ESP32 WIFI Support)
Driver : TMC2209 x4 (use TMC2209 standalone in config)
LCD : MKS mini v3.0 (RBG)

https://youtu.be/odYO6YuWUuE

### Version of Marlin Firmware

bugfix 2.1.x

### Printer model

Kossel Delta

### Electronics

MKS TINYBEE TMC2209 x4

### Add-ons

TMC2209 Standalone MKS TINYBEE (ESP32 WIFI)

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/8987420/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/8987421/Configuration_adv.zip)

[image]

[image]

[image]

thisiskeithb, 2022-06-26:
Since you’re using TMC drivers in standalone mode, did you properly set the driver current using the potentiometer on the driver itself?

flishy2532, 2022-06-26:
am set with multi meter stepper motor is not hot but when am print on 75% cura layer is shift i think some code is bug

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

thisiskeithb, 2022-06-26:
> am set with multi meter stepper motor is not hot but when am print on 75% cura layer is shift i think some code is bug

Try increasing your driver current.

flishy2532, 2022-06-27:
Try increasing your driver current. am increasing but it cant help it layer shift any one can help am try turn on and off all config
alway layer shift

thisiskeithb, 2022-06-27:
> am increasing but it cant help it layer shift 

If your driver current is too low, it can also cause layer shifts.

Also check the grub screens on your motor pulleys and ensure those haven’t come loose.

flishy2532, 2022-06-27:
> > am increasing but it cant help it layer shift 
> 
> If your driver current is too low, it can also cause layer shifts.
> 
> Also check the grub screens on your motor pulleys and ensure those haven’t come loose.

am check all motor pullyes it work 100% 
when am print 1 hours around 70% cura it always layer shift look this my print frist start it good when go half it shift 😭

1.

[image]

2.

[image]

flishy2532, 2022-06-27:
Any one can help fix?

flishy2532, 2022-06-27:
am never give up fix this board i think have bug layershift
am test all function turn on and off 
2 month ago  it not fix 🥺😭🐬

[image]

am lost many filament on this board no care only want fix it 👨‍🔧 

[image]

any one can help ?🤔🥰 👨‍🔧

Symon82, 2022-06-27:
Try to disable PRINTCOUNTER (in Configuration.h) and avoid writing anything into EEPROM (do not "Store Settings" or use M500).

flishy2532, 2022-06-27:
> Try to disable PRINTCOUNTER (in Configuration.h) and avoid writing anything into EEPROM (do not "Store Settings" or use M500) while printing.

let me test

fBn0523, 2022-06-27:
You need to cool the TMC sufficiently
Note: The wrong version of the program will cause wifi to fail to upload files
I think this can help you
[MKS TinyBee V1.0 User Manual.pdf](https://github.com/MarlinFirmware/Marlin/files/8993525/MKS.TinyBee.V1.0.User.Manual.pdf)
.

nikybiasion, 2022-07-02:
I have the same issue wih tinybee and tmc2209, i set vref to 1.25/1.26 for all drivers and put a cooler fan over them, now no more layer shift

flishy2532, 2022-07-02:
layer shift because enable #print counter

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>

closed 2022-07-02 by thisiskeithb

flishy2532, 2022-07-02:
did you enable #define printcounter in config?

รับ Outlook สำหรับ Android<https://aka.ms/AAb9ysg>