MarlinFirmware/Marlin #24416: [BUG] Ender 3 Pro hang than reboot starting second layer

theonethread, 2022-06-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Installing the new firmware from SD card, my prints stop after the first layer, the printer hanging then rebooting.

* Experienced using my personal fork which uses `2.1.x` branch as base with configurations for Ender-3 Pro, then created a new branch of `bugfix-2.1.x` merging my changes, but same outcome.
* Tried to change PrusaSlicer settings, model orientation, same outcome.
* Connecting to my laptop through USB cable Pronterface reported `SD read error` before the restart, so tried a different SD card, but same failure.

I kept my fork up to date compiling after each merge, but haven't updated the firmware on my printer for a time (might be it was running `2.0.x` successfully before), so can not narrow down the timeframe when the issue presented itself.

### Bug Timeline

Found updating the firmware after ca. a month

### Expected behavior

Prints carry on after first layer.

### Actual behavior

Printer errors, hangs, then reboots.

### Steps to Reproduce

1. Build fork [repository](https://github.com/theonethread/Orc-Flavored-Marlin)
2. Install firmware on a machine with same characteristics:
    * board: Creality 4.2.2 32bit
    * cpu: STM32F103 RET6
    * stepper driver: HR4988 (sharpie C on SD slot)
3. Start a job of GCode attached
4. Wait for first layer to finish, then see hang & reboot

### Version of Marlin Firmware

2.1.x latest

### Printer model

Creality Ender-3 Pro v1.5

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Configuration (instead `zip`):
https://github.com/theonethread/Orc-Flavored-Marlin
The point, when the error happens:
[image]
GCode used:
[gcode.zip](https://github.com/MarlinFirmware/Marlin/files/8991253/gcode.zip)
Find configuration in fork repository

matiaschamu, 2022-06-27:
Hello theonethread, I have the same error, when change from first to second layer the printer freeze and reboot

theonethread, 2022-06-28:
Hi @matiaschamu, could you please share your printer setup, which version change did you experience the issue first, and probably your configurations? Those help the maintainers, thanks in advance!

eviltwinskippy, 2022-06-30:
I am also running across this issue, with similar hardware. I have the same Creality 4.2.2 motherboard. My unit is a Ender 3d Pro, with BLTouch, the second z axis motor, and a Capricorn replacement filament tube. Trying to compare behavior with prior builds is, of course, impossible. I had to upgrade the firmware to get the BLTouch to work at all. I am using builds produced by marlin.crc.id.au. (Trying to download software from Creality was ... frustrating.) Currently I am running a nightly build (20220624). I have had the same problem on 20220619, as well as vendor supplied firmwares dated 20210306.

I have observed the same gcode crashing at different points. Small models seem to print fine. But when I go to print anything on the order of 100mm or more across, the machine looses its mind. It may be on the second layer. It may be 12 hours into a 48 hour job. I've tried slicing models with Cura. I've tried slicing models with Prusa. I've tried limiting Jerk. I've tried making the model run faster, or slower. I'm just thinking my machine is cursed.

Prior to the BLTouch/Zaxis upgrades I was getting random thermal runaways at about the same rate (and in the same models) that I'm getting inexplicable reboots now. 

Other odd things that I see with almost every model I print: the darn user interface tends to act like I smacked the button at random.

theonethread, 2022-06-30:
@eviltwinskippy what CPU did your board ship with? it's always harder to identify "random" freezes... :( I had the exact same problem with every approach I took, that is why I posted the report (still might be the problem is with my hardware). you may have a short circuit, might run out of RAM, and million other reassons. can you try to build the firmware yourself / have someone to build it a test for you? would be a good comparison to see what are your findings with current `bugfix-2.1.x`, if possible. hope you find a solution soon!

eviltwinskippy, 2022-06-30:
@theonethread I have the RET6 cpu
[image]

eviltwinskippy, 2022-06-30:
Here is a video of the crash in progress

https://user-images.githubusercontent.com/5579902/176673267-b54b1c49-0a8a-459e-befb-133035f34a89.mp4

And here is the model and the 2 different Gcodes I've generated from 2 different slicers:

CE3PRO - Cura with my jazzed out settings
PRUSA - Prusa with bog standard defaults
SIMPLE - Cura with bog standard defaults
* The STY File
* The OpenSCAD file that generated the STY file

[TestFiles.zip](https://github.com/MarlinFirmware/Marlin/files/9019605/TestFiles.zip)

Basically this is a cube with rounded edges that is hollowed out with a hexagon, with two holes up top to accommodate this print being screwed into another product. (Namely a yarn bowl I am trying to print for my wife that is also crashing, but that monster is a 2 day print and about 275g of filament.)

eviltwinskippy, 2022-06-30:
> You have GD32 (a not fully compatible clone of a STM32 chip, also not supported by Marlin) model F303 also not supported by Marlin.
> 
> Use manufacture provided firmware

You do realize that Creality *ships* Marlin from the factory.

So let's just see my options...

(Downloads from https://www.creality.com/pages/download-ender-3-pro?spm=..page_1934481.products_display_1.1)

The only firmware they offer for download that includes BLTouch is for the 4.2.7 board. Who knows what the "High Temperature Full Hotend Kit" is? When I try to grab the "Ender-3 Pro 4.2.2 mainboard firmware", I get a dodgy zip file that contains firmware for all of the 8 bit boards. And the date on that is from January 2021.

(Angry tirade follows. Most of it directed at the bean counters at Creality. Though, truth be told, if the board wasn't so dang cheap my wife probably wouldn't have bought it for me...)

eviltwinskippy, 2022-06-30:
So to summarize my issue: I have a knockoff processor, and my best/only option to make the beast work reliably is to replace the motherboard. Honestly, I really can't complain too much, I think the original unit only set me back $125. I've blown more than that on filament. By some miracle I did manage to get it to behave long enough to print this:
[image]

But even with that, it crashed after finishing the bottom. I got it to complete by hacking the Gcode, and tricking the printer into "starting" the print on the 29th level. (And tricking the BLTouch into thinking the bottom of the bowl was the zero height of the plate...)

eviltwinskippy, 2022-06-30:
An update: I briefly tried the Creality provided firmware. Either it's completely bonkers, or I'm doing something wrong. The bed leveling system is indicated to be CR Touch, not BL Touch. And something is backwards because trying to use my old settings embedded my nozzle straight into my shiny new glass bed.

I have been running most of the day with the "Frankenfirmware" provided by H5 labs:

https://github.com/MarlinFirmware/Marlin/files/8174702/Config.Files.and.Ender3-Pro.BIN.for.GD32F303.micro.zip

The print that was giving me so much trouble before is now nearly finished after a single try: 

[image]

So if you find yourself in the same boat I was in, a commercially purchased Ender with a v4.2.2 board powered by a "STM32 substitute", and you want to add BL Touch, there is something available.

theonethread, 2022-07-01:
There were a bunch of commits waiting for me today, my `bugfix-2.0.x` branch finally compiles without error, so will first try the updated `2.1.x` derivative, and if the issue still exists, can also try my `2.0.x` derivative, and get back with intel.

theonethread, 2022-07-01:
Well, same issue with both bugfix branches... :( If I'll have time in the weekend, will compile a "vanilla" 2.1 version with no extra features to try out.

theonethread, 2022-07-10:
Had some time this weekend, created a vanilla branch from bugfix-2.1.x + ender 3 config containing only my calibrations and bed tramming, and with that my print finished fine. Not sure, what feature caused the crash (or which features collided?), will slowly add them back, and if I find out more, will let you know.

thisiskeithb, 2022-07-10:
STM32 SDIO fixes were just merged 18 hours ago, so you should probably do another `bugfix-2.1.x` pull if you were working from an earlier version.

theonethread, 2022-07-10:
Thanks, just merged the changes to my vanilla and customized branch, both compile fine, only had a starting warning: "**Warning! Ignore unknown configuration option `monitor_flags` in section [env]**" Will test the firmware on the machine later when I get home.

jmeeter, 2022-07-22:
> An update: I briefly tried the Creality provided firmware. Either it's completely bonkers, or I'm doing something wrong. The bed leveling system is indicated to be CR Touch, not BL Touch. And [...]

Will the attached firmware work with the Ender 3 V2 as well or only the Ender 3 Pro?

Worlixz, 2022-10-03:
Hi there,

I come to you because I have the same concern as the author of the topic.

I tried the two available firmwares but I have the same problem.

Has anyone managed to solve the problem?

I'm also on a Creality Ender-3 Pro v1.5 with the marlin 2.0.9.3 displayed on startup

closed 2022-12-13 by github-actions[bot]