MarlinFirmware/Marlin #24419: [BUG] bugfix 2.0.1 will not build with ender 3 V427 configs

ellensp, 2022-06-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

One example of many.

using stock bugfix 2.0.x configs on bugfix 2.0.x code results in

macro "LOGICAL_AXIS_ARRAY_1" requires 2 arguments, but only 1 

using stock bugfix 2.1.x configs on bugfix 2.1.x code complies as expected

### Bug Timeline

no idea

### Expected behavior

It should build

### Actual behavior

error:

```
Marlin/src/lcd/menu/menu_advanced.cpp:420:34: error: macro "LOGICAL_AXIS_ARRAY_1" requires 2 arguments, but only 1 given
  420 |         LOGICAL_AXIS_ARRAY_1(9999)
      |                                  ^
In file included from Marlin/src/lcd/menu/../../inc/MarlinConfig.h:47,
                 from Marlin/src/lcd/menu/../marlinui.h:24,
                 from Marlin/src/lcd/menu/menu.h:24,
                 from Marlin/src/lcd/menu/menu_item.h:24,
                 from Marlin/src/lcd/menu/menu_advanced.cpp:31:
Marlin/src/lcd/menu/../../inc/../core/types.h:59: note: macro "LOGICAL_AXIS_ARRAY_1" defined here
   59 | #define LOGICAL_AXIS_ARRAY_1(E,V)  { LOGICAL_AXIS_LIST_1(E,V) }
      | 
Marlin/src/lcd/menu/menu_advanced.cpp:463:35: error: macro "LOGICAL_AXIS_ARRAY_1" requires 2 arguments, but only 1 given
  463 |         LOGICAL_AXIS_ARRAY_1(99000)
      |                                   ^
In file included from Marlin/src/lcd/menu/../../inc/MarlinConfig.h:47,
                 from Marlin/src/lcd/menu/../marlinui.h:24,
                 from Marlin/src/lcd/menu/menu.h:24,
                 from Marlin/src/lcd/menu/menu_item.h:24,
                 from Marlin/src/lcd/menu/menu_advanced.cpp:31:
Marlin/src/lcd/menu/../../inc/../core/types.h:59: note: macro "LOGICAL_AXIS_ARRAY_1" defined here
   59 | #define LOGICAL_AXIS_ARRAY_1(E,V)  { LOGICAL_AXIS_LIST_1(E,V) }

```

### Steps to Reproduce

1 buildroot/bin/use_example_configs "Creality/Ender-3/CrealityV422"
2. Autobuild marlin (or update default environment and build) 
3. macro hell breaks out.

### Version of Marlin Firmware

bugfix 2.0.x

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Seen https://reprap.org/forum/read.php?415,889763
and in discoord.

ellensp, 2022-06-28:
On updating the following defines

#define LOGICAL_AXIS_LIST_1(V)  LINEAR_AXIS_LIST_1(V) LIST_ITEM_E(V)
#define LOGICAL_AXIS_ARRAY_1(V)  { LOGICAL_AXIS_LIST_1(V) }

further errors are found

```
Marlin/src/module/../inc/../core/types.h:51:52: error: 'Z_MAX_BED' was not declared in this scope; did you mean 'Y_MAX_BED'?
   51 | #define MAIN_AXIS_NAMES     LINEAR_AXIS_LIST(X, Y, Z, I, J, K)

```

This was traced back to 
  #define _AMAX(A) (_AXIS(A) < Z_AXIS ? A##_MAX_BED : A##_MAX_POS)

Which simply does not work when called in 

MAPLIST(_AMAX, MAIN_AXIS_NAMES)

thinkyhead, 2022-06-29:
I have posted a patch, and will soon follow up with some other patches bringing 2.0.x more in parity with 2.1.x, in prep for a quickie minor release tag later today.

moonglow, 2022-06-29:
@thinkyhead 

Looks like your patch does not fix that macro '#define _AMAX(A) (_AXIS(A) < Z_AXIS ? A##_MAX_BED : A##_MAX_POS)'  it will extend to something like that:

`(Z_AXIS < Z_AXIS ? Z_MAX_BED : Z_MAX_POS)` so `Z_MAX_BED` is must exist anyway. It will not be used but we need it anyway. We can create fake value or add something like that to `Conditionals_post.h`:
```C
#if HAS_Z_AXIS
  #define Z_MAX_BED (Z_MAX_POS) /* prevent compilation error */
#endif
```

ellensp, 2022-06-29:
Confirmed.  Still stick on "Z_MAX_BED"

tom-el, 2022-06-30:
Ran into the LOGICAL_AXIS_ARRAY_1/Z_MAX_BED problem. Tried the 2.1 Configuration examples and the zip file seems to be corrupted. Unless I'm doing something obviously stupid....

ellensp, 2022-07-01:
seems to be resolved with https://github.com/MarlinFirmware/Marlin/commit/088fa84b7f78e1632f74743e0868773b11e1666e

commit 40fa85b9 (2022-06-29): 🐛 Fix types.h macros and fields

Fixes #24419
commit af4160af (2022-07-28): 🐛 Fix types.h macros and fields

Fixes #24419
closed 2022-07-01 by ellensp