MarlinFirmware/Marlin #24422: [BUG] (bug summary)

haitham-alalimi, 2022-06-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I installed the system 2.1 on the mks robin nano 1.2v board and made the necessary changes. The system worked perfectly, but when I inserted the sd card, it rebooted, then I increased and decreased the buadrate, but the benefit and then I did the option SDCARD_CONNECTION onboard and still the same problem.
Then I downloaded bugfix 2.1.x and the result was somewhat better when I insert and remove the sd card several times the files appear but the printing appears distorted and there is an offset when it rises a few layers 
  

### Bug Timeline

New bug

### Expected behavior

Rebot 

### Version of Marlin Firmware

2.1.x

### Printer model

Creasee cs 30 

### Electronics

Mks robin nano v1.2 

### Add-ons

Inductive sensor 

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[image]
[image]

haitham-alalimi, 2022-06-28:
H

haitham-alalimi, 2022-06-28:
Configuration.h and Configuration_adv.h.
[Marlin2.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/9000706/Marlin2.1.x.zip)

haitham-alalimi, 2022-06-28:
Video in this link:
https://drive.google.com/file/d/1y9bpXRxt0bDRF1xGjbJnL9DI53OKZQYH/view?usp=drivesdk

GHGiampy, 2022-07-02:
I see a lot of problems related to the current SD card implementation (at least with Ender 3/V2/Pro) from different opened issues:
- Random reboots
- Random layer shifts
- Random freeze
- Random failures in SDCard initialize

I was a little behind, in "old" bugfix-2.0.x branch (at 2022-02-22) and everything was working fine when I implemented the long filename on FAT and firmware upload via USB. Now, updating to the latest "new" bugfix-2.0.x I experienced problems such as:
- M21 (Init SD card) random init errors.
- M20 (List SD card) sometimes lists nothing.
- M30 (Delete SD file) sometimes reports error and truncate the filesystrem chain, so the remaining files in the directory are deleted.
- M28 (Start SD write) very often is unable to open the file to write to.
- M29 (Stop SD write) not always save the file content.
- Very often the serial output for SDCard commands stops responding.
- The SDCard commands lags and are quite slow to execute, related to the february 2022 version.
- Obviously the upload via USB is entirely broken.

I have seen code moved (watchdog, ...), forced baudrate switch (250000 vs 115200 in ProUI, is this really needed?) and some motors/HW calls into the SDCard functions (isn't a good practice) that could interfere, expecially if these functions share an I2C bus or an interrupt, but I don't know how it works at low level.

I apologize if i'm the only one with all those problems (because i'm also using Neopixel leds) but IMHO the current SDCard implementation is unuseful in this state, too unstable and inconsistent to read gcode files from. A rollback should be made to track down the problems and restore a stable implementation.

Meanwhile I reverted to the 2022-02-22 bugfix-2.0.x "old" branch and everything is working as it was, so this excludes the physical SDCard from the problem.
As soon as I have time I will try the bugfix-2.1.x branch.

mriscoc, 2022-07-08:
To confirm, Marlin BFT was working ok with Bugfix 2.0 and started to fail with Bugfix 2.1, I will try to research and revert the SDIO commit to see if that fixes the bug.

benlye, 2022-07-08:
And to be clear, Binary File Transfer fails because the protocol can't create the new file on the SD card.  

At the same time the `M30` command fails when trying to delete files on the SD card, so it seems like there is an underlying SDIO issue occurring, which is what appears to link the BFT problem to this issue.

GHGiampy, 2022-07-08:
I need to clean up my branches before doing an acceptable PR, meanwhile here is a smart patch as a starting point...
[sdio.zip](https://github.com/MarlinFirmware/Marlin/files/9074385/sdio.zip)

thinkyhead, 2022-07-09:
I went ahead and put up a PR that can be tested with and patched up - link above. If it comes down to needing to have a whole different SDIO for STM32H743 versus others, just to keep things working as before, that's what we'll do.

thisiskeithb, 2022-07-09:
Please test/report findings in https://github.com/MarlinFirmware/Marlin/pull/24470

closed 2022-07-09 by thisiskeithb