MarlinFirmware/Marlin #24426: [BUG] Sanity Check !_IS_5V_TOLERANT(Z_MIN_PROBE_PIN) logic may be incorrect for Creality board.

JMLdev, 2022-06-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

It is impossible to use the #define BLTOUCH_SET_5V_MODE feature with #define Z_MIN_PROBE_PIN on the Creality board 4.2.2, but I think that the default BL_T plug is 5v compatible. [This pinout diagram](https://user-images.githubusercontent.com/242382/146306819-e6d17d4f-3978-41bd-bb7a-1ef692daf454.png) shows that the power rail is 5v. There is no default option that I see within the configuration.h file to define that plug, so the Z_MIN_PROBE_PIN option must be used with PB1 as the sensor, as far as I can tell. Is the logic in `Marlin\src\inc\SanityCheck.h` wrong, or am in incorrect about using BLTOUCH_SET_5V_MODE  for BLtouch v3.1 with the BL_T 5 pin plug?

### Bug Timeline

There seems to be several old topics about this, none of which were resolved. Most closed to inactivity. 

### Expected behavior

I expected to be able to compile with support for BLtouch v3.1 in 5v mode.

### Actual behavior

With `BLTOUCH_SET_5V_MODE` enabled, error: `#error "BLTOUCH_SET_5V_MODE is not compatible with the Z_MIN_PROBE_PIN."` is thrown. The offending line is 1674 in SanityCheck.h and the defined 5v support is on line 1667. THe full code block is:

```
// BLTouch can't run in 5V mode with a 3.3V probe pin
    #if ENABLED(BLTOUCH_SET_5V_MODE)
      #define _5V(P,A,B) WITHIN(P,A,B)
      #ifdef STM32F1            // STM32F103 5V-tolerant pins
        #define _IS_5V_TOLERANT(P) (_5V(P,PA8,PA15) || _5V(P,PB2,PB15) || _5V(P,PC6,PC12) || _5V(P,PD0,PD15) || _5V(P,PE0,PE15) || _5V(P,PF0,PF5) || _5V(P,PF11,PF15))
      #elif defined(ARDUINO_ARCH_SAM)
        #define _IS_5V_TOLERANT(P) 0 // Assume no 5V tolerance
      #else
        #define _IS_5V_TOLERANT(P) 1 // Assume 5V tolerance
      #endif
      #if USES_Z_MIN_PROBE_PIN
        #if !_IS_5V_TOLERANT(Z_MIN_PROBE_PIN)
          #error "BLTOUCH_SET_5V_MODE is not compatible with the Z_MIN_PROBE_PIN."
        #endif
      #elif !_IS_5V_TOLERANT(Z_MIN_PIN)
        #if !MB(CHITU3D_V6)
          #error "BLTOUCH_SET_5V_MODE is not compatible with the Z_MIN_PIN."
        #endif
      #endif
      #undef _IS_5V_TOLERANT
      #undef _5V
    #elif NONE(ONBOARD_ENDSTOPPULLUPS, ENDSTOPPULLUPS, ENDSTOPPULLUP_ZMIN, ENDSTOPPULLUP_ZMIN_PROBE)
      #if USES_Z_MIN_PROBE_PIN
        #error "BLTOUCH on Z_MIN_PROBE_PIN requires ENDSTOPPULLUP_ZMIN_PROBE, ENDSTOPPULLUPS, or BLTOUCH_SET_5V_MODE."
      #else
        #error "BLTOUCH on Z_MIN_PIN requires ENDSTOPPULLUP_ZMIN, ENDSTOPPULLUPS, or BLTOUCH_SET_5V_MODE."
      #endif
    #endif
```

### Steps to Reproduce

1. uncomment #define Z_MIN_PROBE_PIN PB1 in Configuration.h
2. uncomment #define BLTOUCH_SET_5V_MODE in Configuration_adv.h
3. set any other related BLtouch settings
4. Try to compile, observe the error.Bu

### Version of Marlin Firmware

Marlin-Bugfix-2.1.x

### Printer model

Creality Ender 5 Pro

### Add-ons

Micro-swiss direct drive, BLtouch 3.1

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

There doesn't seem to be a spot to upload a zip file with the configuration. I have an [open pull request into the config repo here](https://github.com/MarlinFirmware/Configurations/pull/759). Just make the changes to the Configuration.h and Configuration_adv.h files that I listed in my repro steps.

thisiskeithb, 2022-06-29:
> This pinout diagram shows that the power rail is 5v.

The rail might be 5V, but the MCU pins are not. This is not a bug.

> I have an open pull request into the config repo here. 

`BLTOUCH_SET_5V_MODE` should be disabled in those configs.

closed 2022-06-29 by thisiskeithb

JMLdev, 2022-06-29:
@thisiskeithb okay, thank you for confirming. I spent a lot of time trying to figure this out and it was difficult without proper documentation for the 4.2.2 boards. I appreciate the reply. `BLTOUCH_SET_5V_MODE` is disabled in the pull request. You'll notice in the Additional information section above, I indicated that changes would have to be made to the PR files to repro. Cheers.