MarlinFirmware/Marlin #24438: [BUG] Power_loss fail

FreeZuu69, 2022-07-01:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

After a power cut, the printer resumes from where it left off, and the house immediately goes to the xy position without any heating. For this reason, the nozzle sticking with the melting filament drags the piece from the table.

### Bug Timeline

new

### Expected behavior

as the printer resumes after a power outage, I would like the desk and head to warm up before I go home.

### Actual behavior

It makes xy home without heating the nozzle and table.

### Steps to Reproduce

power_loss ( yes ) 
nozzle and bed heating
go to xy home
last start print

### Version of Marlin Firmware

Marlin-2.0.9.3

### Printer model

Corexy

### Electronics

Mks S gen L 2.0 , Mks Pwc, Mks H43 LCD, bltouch

### Bed Leveling

ABL 3-point

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

https://drive.google.com/drive/folders/1sG6nTN0TP-Ce3W0sxNEfWWX1xA_TItUo?usp=sharing

thisiskeithb, 2022-07-01:
Heating happens after homing, so this is a feature request.

FreeZuu69, 2022-07-01:
how can I do that? Can you help me ?

FreeZuu69, 2022-07-02:
https://drive.google.com/drive/folders/1sG6nTN0TP-Ce3W0sxNEfWWX1xA_TItUo?usp=sharing

As can be seen in the video at the address, the system switches to the parking position before the nozzle and table warm up. He needs to warm up first and then go to park position. I've been dealing with this problem for weeks Please help me.

thisiskeithb, 2022-07-02:
>  I've been dealing with this problem for weeks Please help me.

This is a feature request, not a bug, so someone will need to code the options you are looking for.

If you would like this feature immediately, you can post in the [#marlin-devs-for-hire channel on our Discord](https://discord.gg/n5NJ59y) and pay someone to implement this feature, otherwise you will need to wait until someone finds & completes this request.

FreeZuu69, 2022-07-02:
My second machine using the same marlin software keeps warming up before continuing in the power outage. This does not happen with the mks s gen L. Don't you think this is a mistake?