MarlinFirmware/Marlin #24439: Compile Error on bugfix code from 2/7/2022

nicwilson58, 2022-07-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If  #define CR10_STOCKDISPLAY  and  #define RET6_12864_LCD are enabled in the bugfix code from 2nd July 2022 a compile error occurs as follows that did not occur on previous builds.

Linking .pio\build\STM32F103RE_creality\firmware-20220702-150631.elf
c:/users/nicwi/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/STM32F103RE_creality/src/src/lcd/dogm/u8g_fontutf8.cpp.o: in function `fontgroup_drawstring(_font_group_t*, void const*, char const*, unsigned char (*)(unsigned char const*), void*, int (*)(void*, void const*, char const*)) [clone .constprop.0]':
C:\Users\nicwi\Downloads\Marlin-bugfix-2.1.x/Marlin\src\lcd\dogm/u8g_fontutf8.cpp:110: undefined reference to `get_utf8_value_cb(unsigned char const*, unsigned char (*)(unsigned char const*), unsigned long&)'
c:/users/nicwi/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/STM32F103RE_creality/src/src/lcd/lcdprint.cpp.o: in function `lcd_put_u8str_P(char const*, signed char, char const*, __FlashStringHelper const*, unsigned char)':
C:\Users\nicwi\Downloads\Marlin-bugfix-2.1.x/Marlin\src\lcd/lcdprint.cpp:51: undefined reference to `get_utf8_value_cb(unsigned char const*, unsigned char (*)(unsigned char const*), unsigned long&)'
c:/users/nicwi/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/STM32F103RE_creality/src/src/lcd/marlinui.cpp.o: in function `_wrap_string(unsigned char&, unsigned char&, char const*, unsigned char (*)(unsigned char const*), bool)':
C:\Users\nicwi\Downloads\Marlin-bugfix-2.1.x/Marlin\src\lcd/marlinui.cpp:449: undefined reference to `get_utf8_value_cb(unsigned char const*, unsigned char (*)(unsigned char const*), unsigned long&)'
c:/users/nicwi/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\nicwi\Downloads\Marlin-bugfix-2.1.x/Marlin\src\lcd/marlinui.cpp:427: undefined reference to `get_utf8_value_cb(unsigned char const*, unsigned char (*)(unsigned char const*), unsigned long&)'
c:/users/nicwi/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\nicwi\Downloads\Marlin-bugfix-2.1.x/Marlin\src\lcd/marlinui.cpp:438: undefined reference to `get_utf8_value_cb(unsigned char const*, unsigned char (*)(unsigned char const*), unsigned long&)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F103RE_creality\firmware-20220702-150631.elf] Error 1

### Bug Timeline

V2.1.0.0 bug fix

### Expected behavior

compile error free.

### Actual behavior

compile error as shown above

### Steps to Reproduce

use provided config link below, on bugfix 2nd July 2022 and compile

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/9032467/Marlin.zip)

### Version of Marlin Firmware

2.1.0.0 bugfix 2nd July 2022

### Printer model

Creality Ender 6 with 12864 LCD

### Electronics

stock except for 12864 LCD

### Add-ons

12864 LCD,  it is these defines that cause the compile error.

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-07-02:
This started in https://github.com/MarlinFirmware/Marlin/commit/cb4a79173be3e8ed1ff45e7c346df1c28e744d5a and is being worked on.

aystarik, 2022-07-02:
[lchar_tft_fix.txt](https://github.com/MarlinFirmware/Marlin/files/9033652/lchar_tft_fix.txt)

EvilGremlin, 2022-07-02:
@aystarik interesting, for me source of trouble and type mismatch was `pstart`, not `pval`
i thought about changing ch to lch too, but it looks like this name reuse shuldn't affect anything

aystarik, 2022-07-02:
yes, it boots just fine

EvilGremlin, 2022-07-02:
what MCU/display?

aystarik, 2022-07-02:
const uint8_t ch = uint8_t(ch & 0x00FF); -- this was the problematic construction, ch is already defined, but not yet assigned to value, so already used with undefined value

aystarik, 2022-07-02:
mks robin nano v1.2 + 3.5 inch screen

EvilGremlin, 2022-07-02:
> const uint8_t ch = uint8_t(ch & 0x00FF); -- this was the problematic construction, ch is already defined, but not yet assigned to value, so already used with undefined value

ah yep... but was it source of bootloop?

aystarik, 2022-07-02:
for me it was not compiling, and not linking properly.

EvilGremlin, 2022-07-02:
I mashed up this 'fix' but looks like it bootloops on both F1 and F4. #24444

aystarik, 2022-07-02:
I don't see the part for get_utf8_value_cb() problem

EvilGremlin, 2022-07-02:
`return (const char *)get_utf8_value_cb(FTOP(pstart), cb_read_byte, pval);` did it for color_ui

aystarik, 2022-07-02:
As you noticed, real problem is pval -- it was pointer but became reference, thus linking is not possible.

EvilGremlin, 2022-07-02:
OMG i'm totally blind... didn't notice mismatch between .cpp and.h
But linker was showing mismatch as char/undigned char for pstart.
Will you sbmit it as PR?

aystarik, 2022-07-02:
May I just say that I "Signed-off-by: Alexey Starikovskiy <aystarik@gmail.com>" and be done with it?

EvilGremlin, 2022-07-02:
Eh, even that isn't really that necessary :) Scott pull stuff from vendor not-forks al the time :) 
I will mention you if i end up tidying it all up as PR tomorrow.

thisiskeithb, 2022-07-03:
This was fixed in https://github.com/MarlinFirmware/Marlin/commit/2b6ce3006ecad9573076e5fbac7b2a8dd03e74e7

closed 2022-07-03 by thisiskeithb