MarlinFirmware/Marlin #24442: [BUG] TFT_LVGL_UI all pics No data blocks for MKS robin nano V1.2 & MKS_ROBIN_TFT35 despite RGB565 images present(tryed mks_pic/mks_font/assets)

sunnydrake, 2022-07-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

printer Creasee CS30 
compile config thru PlatformIO/VSCode with options
#define MKS_ROBIN_TFT35
#define TFT_LVGL_UI 1
default_envs = mks_robin_nano35
copy all files to sdcard assets* Robin*.bin , make folders mks_font and mks_pic + copy corresponding bin files there
checked bin file with https://rawpixels.net/ width:117  height:88 images is there
flashed and screen is .... 
[image]

Try marlin 2.0/2.1 branches bug still persist

### Bug Timeline

2.0 branch

### Expected behavior

icons shown

### Actual behavior

text No Data

### Steps to Reproduce

1.compile with
#define MKS_ROBIN_TFT35
#define TFT_LVGL_UI 1
default_envs = mks_robin_nano35
2.flash via sdcard
3.observe screen

### Version of Marlin Firmware

2.1. bugfix branch

### Printer model

Creasee CS30

### Electronics

stock,robin nano V1.2 it seems

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9033400/Configuration.zip)

mrnrod45, 2022-07-04:
I have the Same Issue as this upon flashing to my Elegoo Neptune 2. it is the same board type mks_robin_nano35

Sergey1560, 2022-07-06:
The problem is related to the fact that the sd card is not mounted. Therefore, images are not copied.

sunnydrake, 2022-07-06:
i tried long mounting sdcard option from config - it's same :(

rhapsodyv, 2022-07-09:
This board is sdio, so maybe it’s related with our recent sdio issue.
Take a look in the ongoing PR: #24470

thisiskeithb, 2022-07-09:
https://github.com/MarlinFirmware/Marlin/pull/24470 has been merged, so please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

mrnrod45, 2022-07-16:
I tested with bugfix-2.1.x and still no icons. My Printer still boot loops when SD is inserted as well

rhapsodyv, 2022-07-16:
I just tested lastest bugfix-2.1.x and it fully worked. It loaded the assets, boot, no freeze, all icons, no boot loop.

[image]

rhapsodyv, 2022-07-16:
Try another sd or formatting this one.

closed 2022-07-16 by thisiskeithb