MarlinFirmware/Marlin #24445: [BUG] Blank menu screens on DWIN screen

Farva42, 2022-07-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Bugfix-2.1.x as of July 2 2300 MST compiles with a warning when using #define DWIN_MARLINUI_PORTRAIT  
```
Compiling .pio\build\STM32F103RC_creality\src\src\lcd\marlinui.cpp.o
Marlin\src\lcd\e3v2\marlinui\dwin_string.cpp: In static member function 'static void DWIN_String::add(const char*, int8_t, const char*, FSTR_P)':
Marlin\src\lcd\e3v2\marlinui\dwin_string.cpp:60:5: warning: 'ch' may be used uninitialized in this function [-Wmaybe-uninitialized]
   60 |     if (ch == '=' || ch == '~' || ch == '*') {
        |     ^~
```
[config.zip](https://github.com/MarlinFirmware/Marlin/files/9034124/config.zip)

Status screen displays correctly but all menus are missing labels and only show the selection box. Menus can still be selected 
[image]
[image]

### Bug Timeline

New bug occuring after Remove poison wchar_t macro commit

### Expected behavior

Should be able to see menu labels.

### Actual behavior

menu labels do not show.

### Steps to Reproduce

Compile with creality 4.2.2 board, Marlinui dwin portrait.

### Version of Marlin Firmware

Bugfix2.1.x

### Printer model

Ender 3v2

### Electronics

Stock board 4.2.2 with TMC2208 in standalone

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-07-03:
@thinkyhead: Those recent commits are still causing issues with various LCDs/UIs.

thisiskeithb, 2022-07-04:
This also affects `TFT_COLOR_UI`, probably others.

thisiskeithb, 2022-07-04:
Fixed in https://github.com/MarlinFirmware/Marlin/commit/38e9e52fb658e9c3bc50b4535b52ce3ac292693a

closed 2022-07-04 by thisiskeithb

Farva42, 2022-07-04:
Can confirm commit 38e9e52 resolved this issue.