MarlinFirmware/Marlin #24447: Marlin 2.1 Sensorless probing] (While building task, Errors in probe.cpp)

elongmugs, 2022-07-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I went to #define SENSORLESS_PROBING and this error comes while bulding task.
[image]

### Bug Timeline

new bug 03-07-2022

### Expected behavior

I expected to be able to build task and use the sensorless probing just as in marlin 2.0

### Actual behavior

It makes error about probe.cpp

### Steps to Reproduce

1. remove the "//" from #define SENSORLESS_PROBING
2. remove the "//" from #define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2209
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9035450/Configuration.zip)

### Version of Marlin Firmware

2.1

### Printer model

FLsun G

### Electronics

BIGTREE_SKR_PRO v1.1, TMC 2209

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9035452/Configuration.zip)
[image]

Bob-the-Kuhn, 2022-07-04:
Confirmed (with a wrinkle)

As best I can tell there was an attempt to add SENSORLESS_PROBING to Delta printers but a couple of the variables/structures are no longer properly defined for non-Delta printers.  **test_sensitivity** and **offset_sensorless_adj** are two I've run into so far.

I used Auto Build Marlin within VSCode to build using the last set of configuration files attached to the original post.

The first error I get is:
```
Marlin\src\module\probe.cpp:106:68: error: too many initializers for 'Probe::sense_bool_t'
  106 |   Probe::sense_bool_t Probe::test_sensitivity = { true, true, true };
      |                                                                    ^
```

This is immediately followed by the one shown in the author's screenshot.

Bob-the-Kuhn, 2022-07-05:
Looks like the changes are part of PNR [#24220](https://github.com/MarlinFirmware/Marlin/pull/24220#:~:text=G28%2C%20G33%2C%20M48%E2%80%9D-,%2324220,-Merged).

Bob-the-Kuhn, 2022-07-05:
Try using these files.  They at least compile.

[module.zip](https://github.com/MarlinFirmware/Marlin/files/9050203/module.zip)

I have not tried compiling for Delta sensorless homing
.

Bob-the-Kuhn, 2022-07-05:
Updated ZIP of files I'd like tested.
[module2.zip](https://github.com/MarlinFirmware/Marlin/files/9050240/module2.zip)

thisiskeithb, 2022-07-06:
Closing since there’s now a PR: https://github.com/MarlinFirmware/Marlin/pull/24455

closed 2022-07-06 by thisiskeithb

Bob-the-Kuhn, 2022-07-06:
PR has been merged.  Please try the latest bugfix-2.1.x.

elongmugs, 2022-07-07:
Good job! I can use the sensorless probing now! Thx