MarlinFirmware/Marlin #24448: [FR] More optimized ASSISTED_TRAMMING

sbasmanov, 2022-07-03:
### Is your feature request related to a problem? Please describe.

Existing ASSISTED_TRAMMING feature is very useful (especially with BLTouch) in cases, when bed often needs to be leveled after few heat/cooldown cycles or after pulling print from table when it still hot. But the problem is: it have four menu items (corner names) and inside each submenu is only two options - measure and done. When I doing leveling on my stock ender 3 display, it takes a lot of clicks back and forth - entering each menu & exit back. I believe this is not optimal.

### Describe the feature you want

My idea is to have single menu with corner names on left and Z values on right side (similar to edit menu items). And click on menu item will execute measure action without entering submenu. So, it will look like this:
| Front-Left____________00.00 |
| Front-Right___________00.10 |
| Back-Left___________-00.05 |
| Front-Right__________-00.10 |
| Probe all__________________ |
| Done_____________________ |

Also, would be useful to have "Probe all" menu item, to probe all corners at once. Good way to understand if I need leveling at all and if needed - what corners.

Looking at marlin code, I think there must be a way to call ACTION_ITEM with function name and Z value so it will be formatted on screen similar to STATIC_ITEM. But it seems such thing doesn't exist in marlin and I'm not that good at UI code to understand how to make it. In general, it must be something like ACTION_ITEM(MSG,function,value). So, it would be great if someone with good UI skills can help me with creating such menu items (text on left, value on right, click executes action).
Another thing that I'm not sure about - reference Z for leveling. If I start with front left corner it measures to zero (becomes ref point). Rest is measured against this level. I noticed, that at least on my E3 when I adjust other corners, FL corner changes too (a little). So, what happen: after I finished all corners, I measure front left and reference Z is being reset to new value. Of course other corners change their height (because ref changed) and instead of adjusting front left corner, I have to adjust all other corners again (and again and again). So, may be setting new reference Z must me separate option and all subsequent measurements of front left corner will show difference with initial (stored) reference?

Farva42, 2022-07-03:
Have you tried using Level_bed_corners. It sounds like it should do what you are looking for. It will probe the LF(or whatever you set as starting) that sets zero, then moves to RF and will lower z to that zero watching for a probe trigger. If the probe doesn't trigger it prompts to raise the bed till the probe triggers. It will then reprobe to verify then moves to the next corner. If a corner is raised above the set zero that corner is set as the new zero and will loop till all corners are within the tolerance set.

sbasmanov, 2022-07-03:
> Have you tried using Level_bed_corners.
Yes, I tried that. Can't remember now, but I had some problems with it (may be about year ago). So, may be I have to try it again in 2.1 and it will work as expected. But anyway, I would like to make ASSISTED_TRAMMING more easy to use as well :)

sbasmanov, 2022-07-04:
>If the probe doesn't trigger it prompts to raise the bed till the probe triggers.

Ah, I remember now. This is the problem. It finds highest corner and then ask to rise other corners to this height. This is because bltouch can trigger only when tip is moving up. At least on Ender3, rising bed means loosen bed screw (and spring). So, slowly, bed is rising each time :) Finally, I found that one of springs was too loose. That's why I use assisted tramming - it shows me Z difference and adjustment direction.

Andy-Big, 2022-07-12:
I agree - the ASSISTED_TRAMMING menu is not very convenient, you have to press many times to go to the next corner. And I also agree about the zero value of the reference corner - it would be much more convenient if, in addition to the relative value of the height of the corners, the absolute value (of course, taking into account the Z-offset value) would be indicated so that the reference corner could be adjusted to zero height.
I also saw the definition of the thread type in the source code, but it was never used in the wizard to tell in which direction and how much to turn the screw.

sbasmanov, 2022-07-12:
Last few days I printed with 0.2 nozzle and 0.1 layer, so I had to adjust bed exactly to 0.00. And this is pain :(
Biggest problem - front left corner automatically becomes new reference (zero) after measurement. Very common situation - FL 0, FR 0.05, BR 0.05, BL 0.05. So, in this situation much easier to adjust FL by 0.02 and then another corners become leveled with FL. But at this moment it's not possible to know real value of FL, because it always measure as 0.00.
So, in addition to original idea, it seems to me there must be some reference point (coordinates) defined and separate menu item "Set new reference Z". This will measure Z at this point and set it as new zero level for all corners. This way, FL becomes just corner with its own difference from zero and can be easily adjusted.

So, the only problem for me is: how to print menu line same as edit value menu, but on click it will call my function instead of opening submenu or editing value. Looks like I have to call somehow MenuEditItemBase::draw from MenuItem_function::draw (which have to defined somewhere?). All those ++ classes and defines driving me crazy :)

Andy-Big, 2022-07-13:
I will still display both absolute and relative heights :) This will make it possible to align the corners not only relative to each other, but also align them to zero height relative to the nozzle.

sbasmanov, 2022-07-13:
I agree, in some situations absolute height can be useful. In my case it is useless, because bed center always lower than corners :) Also, on stock Ender 3 display there is not much space. So, may be on TFT it will look good, but on my display it looks really terrible :) Anyway, I think most important part is changing how reference Z works. With my changes, now I can check all corners and select any of them as reference (it will just use its height) and if I measure this corner later, it will not reset reference height.
So, I tried different ways to display information in menu and realized that I absolutely don't understand how Marlin UI works. Now my menu looks like this:
[image]

Andy-Big, 2022-07-13:
For half a year now, I have been swearing every time I have to do something in the interface that is at least a little different from the meager standard elements :)))

sbasmanov, 2022-07-14:
Ok, I give up :) I don't understand how menu works and how to build it properly. I made changes to original file, so now it shows absolute (A:) and relative (R:) values for each corner. Reference height can be changed in corner menu (Set new Ref Z). Looks terrible, I know, but that's all I can do with UI :( If someone will make it better - would be great.
[menu_tramming.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/9113259/menu_tramming.cpp.zip)

mikemerryguy, 2022-10-17:
i use the probe assist, in order to not keep loosening the bed you need to run it and skip the high corner, then tighten it down. if you pay attention it shows last z and you can see the differance.  so tighten that corner down and hit skip and it will continue around the bed and allow you to then keep adjusting until all 4 are even.  it is kind of a pain, but it does work very well

thisiskeithb, 2022-10-17:
Using a center point first also helps:

(From an Ender-3)
```cpp
  // Define positions for probing points, use the hotend as reference not the sensor.
  #define TRAMMING_POINT_XY { {X_CENTER, Y_CENTER }, { 32, 32 }, { 203,  32 }, { 203, 203 }, { 32, 203 } }

  // Define position names for probe points.
  #define TRAMMING_POINT_NAME_1 "Center"
  #define TRAMMING_POINT_NAME_2 "Front-Left"
  #define TRAMMING_POINT_NAME_3 "Front-Right"
  #define TRAMMING_POINT_NAME_4 "Back-Right"
  #define TRAMMING_POINT_NAME_5 "Back-Left"
```

LMBernardo, 2022-12-18:
> Very common situation - FL 0, FR 0.05, BR 0.05, BL 0.05. So, in this situation much easier to adjust FL by 0.02 and then another corners become leveled with FL.

What's stopping you from adjusting the FL instead of the other corners?

sbasmanov, 2022-12-18:
>Using a center point first also helps:

Hmm.. Thats's good point. Never tried it. My bed center little higher (about 0.1) than corners, so I never used it for leveling.

> What's stopping you from adjusting the FL instead of the other corners?

By default, when only four points configured, this wizard takes first (usually FL) point as zero reference. All other points shown against this one. And when adjusting (with screw) one corner, another corners change their level too. So, using relative values becomes endless adjusting. I'm using patch I attached above for a while and find it more useful (except UI) than original wizard. Now I level bed using absolute values (where zero is real zero, measured during initial home XYZ) only. Much quicker. Only thing missing is user friendly UI and optional initial all corners measurement.

lukasradek, 2023-01-26:
Hello there, I share the love of assisted tramming and I will try to look into this 🙂.
For now I added (pull request) the support for up to 9 tramming points for people with larger beds 🙂.

lukasradek, 2023-01-28:
@sbasmanov Do you see any reason (if so... which?) to still have absolute tramming values in case that the reference height does not change itself when re-measuring the reference point?

sbasmanov, 2023-01-28:
>in case that the reference height does not change itself when re-measuring the reference point?

Hmm.. Probably this is same. Only question what to show - absolute or relative. I would make it configurable. Since I made this small patch, I never used option to change reference height - it's just not needed. Using zero after HomeXYZ works very well, because all we need is to have all corners at same level, regardless of reference height. Even if entire table will change its level during adjustment, next print job will make HomeXYZ again and Z will be properly calibrated.
Also, what I think would be great in addition, is to have an option to check all corners at once and to get rid of sub menus for each corner - this is pain on Ender 3 to enter, measure & exit.

lukasradek, 2023-01-31:
Yup.... I am making it quite configurable inside the menu itself.

The menu items will show the tram point name with the measurement on the same line.

User will be able to toggle "Menu mode" between "Measure" and "Point detail".
When in Measure mode, clicking the tram point menu item will trigger measurement.
When in Point detail mode, clicking the tram point menu item will open submenu with details about the selected point.

User will also be able to toggle between milimeters and Mx screw turns (set in Configuration_adv.h).

There is also a "Measure All" button. (I am running a 9 tramming point setup, so that is sorely needed 😁.)

I understand that the Z value is somewhat floating since it depends on homing, but I still let the user pick the point where the reference height is measured, so that the "0" can be somewhere, where the bed already is.

How does it work with absolute values for you? When you Home axes and then measure the homing point in tramming, does it show 0 or the Z Probe Offset?

sbasmanov, 2023-02-01:
>When in Measure mode, clicking the tram point menu item will trigger measurement.

This is amazing! Exactly what I thought.

>I understand that the Z value is somewhat floating since it depends on homing, but I still let the user pick the point where the reference height is measured, so that the "0" can be somewhere, where the bed already is.

Sure, it's always good idea to have choice. May be somebody feels more comfortable adjusting table against one corner. 

>How does it work with absolute values for you? When you Home axes and then measure the homing point in tramming, does it show 0 or the Z Probe Offset?

I don't have center point included, so after homing, I start measuring each corner and I look at absolute values. For my table I found that it levels at 0.07 (absolute). Usually corners show values from 0.06 to 0.1 when I start wizard. So, I just checked, after leveling all corners to 0.07, I started tramming wizard again and it showed that all corners still at 0.07. I think if I will include center point too, it will show zero when measured. 
Z probe offset used to compensate difference between toggle point of bltouch and actual nozzle height. And if I understand properly, Z is always used with this difference already applied.

lukasradek, 2023-02-01:
I do not think it is. Since Z=0 must be when nozzle is touching the bed, that would mean that during your tramming your nozzle would be almost touching the bed... which might be the case, but I think it is not 😁. My guess is that absolute measurement of 0 is with reference to the homing position - when probe triggers, that is your absolute zero... for tramming purposes. But I might be wrong... I am testing it on Marlin Simulator, where probing always returns zero 😁 and my tinker printer is currently being upgraded so i cannot test it there 🙂.

Anyway it is almost done. I will soon submit a pull request and I will let this thread know.

sbasmanov, 2023-02-01:
Hmm.. In simulator it may be different. But real bltouch is set 2-3mm higher than nozzle. When probe moves down with pin deployed, it touching surface and pin start moving up. Inside bltouch there is hall sensor and small magnet on other side of pin. So, when this magnet reach hall sensor, interrupt is triggered. Next, we setup Z offset - difference between bltouch trigger point and nozzle. When probing, nozzle never reaching table (about 1-2mm). So, when homing, after bltouch triggered, printer knows, that current Z is Zoffset higher than nozzle (distance between nozzle and table equals Z offset). Internally, marlin sets Z = Z offset :) and all other movements is made according to this height. This is funny, but with bltouch there is no real zero Z (like with endstop).  When bltouch triggers - we're at Z offset height.

sbasmanov, 2023-02-01:
Anyway. Look at function probe_single_point() in menu_tramming.cpp. A call to probe_at_point always returns real Z value. All relative values are calculated at _menu_single_probe: ftostr42_52(z_reference - z_measured[tram_index]). So, there is no need to think about Z offset and all other related things :) You call probe_at_point and receive real Z height. One more advantage of showing absolute values - it's same sign as Z. So, positive Z value means above table and negative - below table. When showing relative value with z_reference==0, we get opposite results: when nozzle above table it is negative and positive when nozzle below table :) Yes, in this case relative value shows direction where to adjust table: negative - down, positive - up. But I think it's little confusing (at least for me :) May be I'm wrong.

lukasradek, 2023-02-01:
Yes, it usually is super confusing 😁. That is what I am trying to get rid of as well. I actually flipped the calculation so that `diff = measured_z - reference_z`, which means that positive values will always mean "the point, you are measuring is above the reference" and vice versa. This makes sense to me because it is just cartesian system... positive is up.

And if you switch units to screw turns, it will just tell you how much and if counter- or clockwise. It is based on what you define in Configuration_adv.h, where I made it also very straightforward... basically you tell it screw diameter and if turning clockwise lowers or rises the bed.

lukasradek, 2023-02-04:
@sbasmanov
It is DONE... and it is IN.
Let's hope they like it a merge it.

thisiskeithb, 2023-02-04:
Closing since there’s now a PR:

- https://github.com/MarlinFirmware/Marlin/pull/25339

closed 2023-02-04 by thisiskeithb