MarlinFirmware/Marlin #24450: [BUG] Tracked down STM32F1 SD Card problems

GHGiampy, 2022-07-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

As noticed here https://github.com/MarlinFirmware/Marlin/issues/24422#issuecomment-1172930414 , different issues related to the SD Card was spotted in the last days. I found that the problems start from this commit eda61a2cbd1de39001193ca6ebf5c5c22df3878c with the `sdio.cpp` file cleanup (or better a rewrite).
I was able to strip down this commit and the related bootloop fix (c9b97b8d641b974de3023dabf7fedbfa352c232e) and build the tip of bugfix-2.1.x with the resumed SD Card functionalities.
I'm a completely noob in DMA management, but from the code I can see that the new implementation never switches the direction from MEMORY_TO_PERIPH, while the old implementation does the switch based on a read or write operation.
I suspect that this could be the problem or a starting point to look at.
It is worth noting that some people experienced layer shifts, random freeze/reboot and print breaks when using (read) SD Card, so this could be related.
This rewrite should be reviewed from someone with STM32F1 skilled in DMA management, which I'm not, or at least reverted back for the STM32F1 part due to high risk of problems (like power loss recovery which writes to teh SD card).

### Bug Timeline

eda61a2cbd1de39001193ca6ebf5c5c22df3878c

### Expected behavior

To write to the SD Card and have a reliable filesystem (also for read).

### Actual behavior

Can't write to SD Card and as soon as try to write to the SD Card, this silently fails without response, then any further serial/sd card Mxx command are ignored and the FAT table is truncated.
In this stage the only operation that one can do to resume the SD Card is a power cycle.

### Steps to Reproduce

1. Open serial monitor
2. Issue `M28 testfile.gcode` to write a file on SDCard.
3. An "echo:Now fresh file: testfile.gcode" is expected, but the serial communication stops responding with the problems stated before.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender 3 V2

### Electronics

Stock V4.2.2

### Add-ons

Neopixel leds

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

`LONG_FILENAME_WRITE_SUPPORT`, `BINARY_FILE_TRANSFER` and `CUSTOM_FIRMWARE_UPLOAD` enabled

thisiskeithb, 2022-07-04:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/24422

thisiskeithb, 2022-07-04:
Please submit a PR so these fixes can be reviewed.

closed 2022-07-04 by thisiskeithb

rhapsodyv, 2022-07-04:
I remember that about an year ago I fixed all sdio issues in stm32, tested it with lots of different boards and setups.
The work I did with the SDIO in the stm32 marked the widely adoption of this Hal in marlin, because the sd issues was something that was holding too many boards to use it…

I remember that the sdio was unreliable, randomly halting or stoping to read.

Now we have this complete rewrite of sdio code. I don’t know if it was widely tested, but the proper fix probably will demand lots of test to get everything stable again.

GHGiampy, 2022-07-04:
> Duplicate of #24422

This is not a duplicate, the problem arises from a M28 attempt the fails every time.
The other issue was referenced only to avoid a duplicate comment, that I dropped there.

thisiskeithb, 2022-07-04:
We don't need multiple open issues for the same "SD card issues". Leaving a comment in the other issue is perfectly fine.

GHGiampy, 2022-07-04:
> Now we have this complete rewrite of sdio code. I don’t know if it was widely tested, but the proper fix probably will demand lots of test to get everything stable again.

That's what I'm thinking, too. A complete rewrite of delicate core part in a platform that can't be tested (as stated from the author of the PR) shouldn't be done blinded.

GHGiampy, 2022-07-04:
> We don't need multiple open issues for the same "SD card issues". Leaving a comment in the other issue is perfectly fine.

I opened this new issue to give more info and the bug timeline.