MarlinFirmware/Marlin #24469: [Sensorless Probing - Auto Bed Leveling: BILINEAR (disabled) ] (While using G29 ,errors occur (SLOW Probe fail! - Triggered early.) and (Probing Failed)

elongmugs, 2022-07-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I was trying to use the Auto Bed Leveling: BILINEAR, but it makes error after doing a G29. G28 is working with sensorless homing Z. But G29 does not work with the sensorless probe.

>>> G29  X130.00 Y165.00 Z5.00
Machine Type: CoreCartesian
Probe:
Probe Offset X0.00 Y0.00 Z0.00 (Aligned With & Same Z as Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
  current_position= X130.00 Y165.00 Z5.00 : Probe::set_deployed
deploy: 1
Probe::do_z_raise(5.00)
>>> do_blocking_move_to  X130.00 Y165.00 Z5.00
>  X130.00 Y165.00 Z5.00
<<< do_blocking_move_to  X130.00 Y165.00 Z5.00
>>> do_blocking_move_to  X130.00 Y165.00 Z5.00
>  X130.00 Y165.00 Z5.00
<<< do_blocking_move_to  X130.00 Y165.00 Z5.00
reset_bed_level
>>> Probe::probe_at_point  X130.00 Y165.00 Z5.00
...(50.00, 50.00, raise, 0, probe_relative)
  current_position= X130.00 Y165.00 Z5.00 :
>>> do_blocking_move_to  X130.00 Y165.00 Z5.00
>  X50.00 Y50.00 Z5.00
echo:busy: processing
<<< do_blocking_move_to  X50.00 Y50.00 Z5.00
  current_position= X50.00 Y50.00 Z5.00 : Probe::set_deployed
deploy: 1
>>> Probe::run_z_probe  X50.00 Y50.00 Z5.00
>>> Probe::probe_down_to_z  X50.00 Y50.00 Z5.00
>>> do_blocking_move_to  X50.00 Y50.00 Z5.00
>  X50.00 Y50.00 Z-1.00
<<< do_blocking_move_to  X50.00 Y50.00 Z-1.00
  current_position= X50.00 Y50.00 Z5.00 : sync_plan_position
<<< Probe::probe_down_to_z  X50.00 Y50.00 Z5.00
<<< Probe::run_z_probe  X50.00 Y50.00 Z5.00
>>> do_blocking_move_to  X50.00 Y50.00 Z5.00
>  X50.00 Y50.00 Z10.00
echo:busy: processing
echo:busy: processing
<<< do_blocking_move_to  X50.00 Y50.00 Z10.00
measured_z: 5.00
<<< Probe::probe_at_point  X50.00 Y50.00 Z10.00
>>> Probe::probe_at_point  X50.00 Y50.00 Z10.00
...(130.00, 50.00, raise, 0, probe_relative)
  current_position= X50.00 Y50.00 Z10.00 :
>>> do_blocking_move_to  X50.00 Y50.00 Z10.00
>  X130.00 Y50.00 Z10.00
echo:busy: processing
<<< do_blocking_move_to  X130.00 Y50.00 Z10.00
  current_position= X130.00 Y50.00 Z10.00 : Probe::set_deployed
deploy: 1
>>> Probe::run_z_probe  X130.00 Y50.00 Z10.00
>>> Probe::probe_down_to_z  X130.00 Y50.00 Z10.00
>>> do_blocking_move_to  X130.00 Y50.00 Z10.00
>  X130.00 Y50.00 Z-1.00
<<< do_blocking_move_to  X130.00 Y50.00 Z-1.00
  current_position= X130.00 Y50.00 Z10.00 : sync_plan_position
<<< Probe::probe_down_to_z  X130.00 Y50.00 Z10.00
SLOW Probe fail! - Triggered early.
<<< Probe::run_z_probe  X130.00 Y50.00 Z10.00
  current_position= X130.00 Y50.00 Z10.00 : Probe::set_deployed
deploy: 0
Probe::do_z_raise(5.00)
>>> do_blocking_move_to  X130.00 Y50.00 Z10.00
>  X130.00 Y50.00 Z10.00
<<< do_blocking_move_to  X130.00 Y50.00 Z10.00
>>> do_blocking_move_to  X130.00 Y50.00 Z10.00
>  X130.00 Y50.00 Z10.00
<<< do_blocking_move_to  X130.00 Y50.00 Z10.00
Error:Probing Failed
[ERROR] Error:Probing Failed

measured_z: 0.00
<<< Probe::probe_at_point  X130.00 Y50.00 Z10.00
  current_position= X130.00 Y50.00 Z10.00 : Probe::set_deployed
deploy: 0
  current_position= X130.00 Y50.00 Z10.00 : > probing complete
>>> do_blocking_move_to  X130.00 Y50.00 Z10.00
>  X130.00 Y50.00 Z5.00
echo:busy: processing
<<< do_blocking_move_to  X130.00 Y50.00 Z5.00
<<< G29  X130.00 Y50.00 Z5.00

### Bug Timeline

new bug 07-07-2022

### Expected behavior

I expected it to work with the sensorless probing with the Autoleveling Bilinear

### Actual behavior

Errors are coming and there is no probing happening.

### Steps to Reproduce

1. Use Pronterface
2. Send M502 for reset to Eprom
3. Send M500 to save to eprom
4. Send M111 S247 for M111 S247
5. Send G28 for Homing
6. Send G29 for Probing

### Version of Marlin Firmware

2.1

### Printer model

FLsun G

### Electronics

BIGTREE_SKR_PRO v1.1, TMC 2209

### Bed Leveling

ABL Bilinear mesh

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[module2.zip](https://github.com/MarlinFirmware/Marlin/files/9074236/module2.zip)
Files from #24447 for fixing the probe.cpp error.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9074247/Configuration.zip)

Bob-the-Kuhn, 2022-07-08:
The #24447 changes are now in bugfix-2.1.x.  No need to modify those four files.

Is this the first time you've tried sensorless probing or was it working before?  If it was working before, what was the last revision/version that you know worked?

There's a [new post in Rep Rap Firmware - Marlin](https://reprap.org/forum/read.php?415,888456) that is very similar to this issue (sensorless probing fail on coreXY).  Is that you or a different person?

elongmugs, 2022-07-09:
First of all yes thats me the " TreeLeggedHumaniod" from the RepRap Firmware - Marlin.

This is my first time using "Sensorless Probing" but not my first time using "Sensorless Homing", with "Sensorless Homing" ive managed to do some very good reliable settings, witch is can do homing with the Nozzle at a speed that does not damage the nozzle.

Because this is my first time using "Sensorless Probing" ive had a believe that the settings are proportional to the "Sensorless Homing", but there are some options for the "Sensorless Probing" that need adjusting, as you are mentioning in the "RepRap forum" - ( Do you probe at the same feed rate as you home? )
Answer do this is: "Yes". These settings ive set to same speeds as my homing, both Fast and slow ( See picture )
[image]
And yes i changed the #define Z_PROBE_FEEDRATE_SLOW (Z_PROBE_FEEDRATE_FAST) to (4*60)
[image]

[image]
Conslusion to this is. " No this does not affect anything "
Ive guess you are thinking this would affect the "early triggering" but no. Problem not solved.

Ive tried some trial and error attempts for different results, let me post these below.

"All attemps are with the sensitivity 110 witch is the normal settings for Homing. And after g28 I tried setting the sensitivity down to 50, and then tried the G29,but same early trigger, I then tried sensitivity 0, same results, it gets the early trigger or no trigger. "

Attempt 1
[image]
[image]

Attempt 2
[image]
[image]

Attempt 3
[image]
[image]

Attempt 4
[image]
[image]
[image]

Attempt 5
[image]
[image]
[image]

Conclusion to this is that whatever "Offset Nozzle to Probe" or "Clearance of probe deploy" nothing is making the probing movement with Z, it is not measurering what so ever. Changing the Sensitivity has no effect as in "Sensorless Homing". But the closest ive managed to get a grid was: 
(Attempt 4) 
[image]
Where the grid is completet but the is no measurement.

Other things ive tried but no luck are.

1.
"Before deploy/stow pause for user confirmation" - For more control as if there would be some static issues to cool down before deploy"
[image]

2.
" Ive tried Reversing the Probe Z_MIN endstop, with some interesting results it started behaving as a normal probe but offcourse there was no trigger, i did notice tho on my mainboard TMC 2209 stepper drivers RED LED light there was and early trigger while it started moving and another when it hit the nozzle. very interesting. I manged to lower the sensitivity to 50( Normal was 119) and there where no trigger when it started moving. So i tried reversing the Endstop again thinking this would solve the early trigger, but no. there was no affect what so ever.
[image]

3.
Endstop noise treshold but these are not combatible with sensorless probing, this is interesting.
[image]

4.
Turning Software Endstop Off while Probing. (No effect).

Conclusion.
I guess this i what i know so far, ill keep trying different approaches but i start to think this as something to do with the kinematics of the ABL program, this i dont know so much about but its just a thought. well i hope this post was helpful somehow, ill stick around if something comes up.

Regards
TreeeLeggedHumanoid aka 

elongmugs.

thisiskeithb, 2022-07-09:
It looks like your printer uses leadscrews on Z, so sensorless probing is not recommended and will basically never work correctly.

Also:

https://github.com/MarlinFirmware/Marlin/blob/62c0ab072cf163202d100cc381b35fc724037c16/Marlin/Configuration.h#L1377

I’d recommend installing a physical probe.

elongmugs, 2022-07-09:
> It looks like your printer uses leadscrews on Z, so sensorless probing is not recommended and will basically never work correctly.
> 
> Also:
> 
> https://github.com/MarlinFirmware/Marlin/blob/62c0ab072cf163202d100cc381b35fc724037c16/Marlin/Configuration.h#L1377
> 
> I’d recommend installing a physical probe.

1.
Yes my printer is using leadscrews on Z.

2.
I dont care if probing is recommended or not, this works with homing and should also with probing.

3.
Basically will never work of course, these are not basics steps to take, these are more advanced. and im interested in the advanced.

4.
Thank you for your recommendation, but i believe this should be an option regardless of different individually opinions of the safety of peoples equipment and there approach to more advanced and difficult "problem solving".

Please explain to me why there is no option for Machines using leadscrews as the Sensorless Homing seems to be an option.
Explain to me why Sensorless probing does not work with ABL and leadscrews.

thisiskeithb, 2022-07-09:
> I dont care if probing is recommended or not, this works with homing and should also with probing.
> Please explain to me why there is no option for Machines using leadscrews as the Sensorless Homing seems to be an option.
> Explain to me why Sensorless probing does not work with the ABL program.

The multiplied torque from leadscrews using sensorless probing _will and has caused damaged_, which is why that warning exists. If you used a belted Z setup like on a Voron 2.x or deltabots, you may have some success, but sensorless probing with leadscrews **is not recommended**.

descipher, 2022-07-09:
Keith is correct, the lead screw will significantly reduce load on the motor and that makes it nearly impossible to detect the correct level of current change sensor less detection is based on.

elongmugs, 2022-07-09:
> The multiplied torque from leadscrews using sensorless probing _will and has caused damaged_, which is why that warning exists. If you used a belted Z setup like on a Voron 2.x or deltabots, you may [...]

I strongly disagree. With the settings i have for my sensorless homing with lead screws i have not managed to damage anything. Its only a question about the right sensitivity and current to the Stallguard detection, i takes time to find the perfect settings and after that the whole thing works like a charm.

I understand people could damage there equipment if not carefull, but this is not about restricting the options for others because apparantly, you guys have bad experiences with leadscrews and fine tuning.

You keep saying its not recommended, i understand. But where is the option to be using this regardless on own risk?

My question is:
How can i get the option to test this for myself on own risk? How can i get this Sensorless Probing and Sensorless Homing to be proportional?

elongmugs, 2022-07-09:
> Keith is correct, the lead screw will significantly reduce load on the motor and that makes it nearly impossible to detect the correct level of current change sensor less detection is based on.

I strongly disagree, ive managed to do this on Sensorless Homing. its about balancing the Sensitivity and Current also the interpolating up to 256 microsteps does the trick. and without damaging anything on of course low speeds as in 4mm/s whitch is plenty of time for the Z to move.

thisiskeithb, 2022-07-09:
> With the settings i have for my sensorless homing with lead screws i have not managed to damage anything

Did you swap in leadscrews for X & Y instead of using belts?

Keep in mind that sensorless homing & sensorless probing are two different things when you add leadscrews to the mix for the aforementioned reasons. Your bed has to also be extremely rigid, with zero give in the mechanics. Many have tried to tune sensorless probing with leadscrews and have limited/no success.

> You keep saying its not recommended, i understand. But where is the option to be using this regardless on own risk?

You've done that already by enabling `SENSORLESS_PROBING`. Marlin doesn't actively block you at this time.

Again, I'd recommend using a real probe with your particular hardware setup and now this is just a matter of you ignoring all the warnings and finding the correct settings for your hardware.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-07-09 by thisiskeithb

descipher, 2022-07-09:
Well you can certainly submit a PR with working code to add that functionality, however it's not a bug.

elongmugs, 2022-07-09:
Is disagree, this is not me ignoring all the warnings. this is me asking the same questions again and again and you are not answering and just sweeping me away with your recommendations.

Explain to me technically why is the Z not moving when im doing Bed leveling. I use the same Movement speeds for homing and for probing and the Homing works fine. Why does is say " Trigger early" It makes no sense.
[image]

thisiskeithb, 2022-07-09:
This is not a support forum. Please use the links for configuration troubleshooting that I listed in [my last comment.](https://github.com/MarlinFirmware/Marlin/issues/24469#issuecomment-1179542441)

elongmugs, 2022-07-09:
no, this is a bug.

elongmugs, 2022-07-09:
The sensorless probing is not working and its a bug.

elongmugs, 2022-07-10:
Sensorless homing and Probing now working.
I kept getting the Probe triggered early. And this Is what is found.

After thinking that sensorless homing and sensorless probing should be using the same Z_min_endstop its doesn’t. ive sent a M111 S247 to pronterface and then sent a M119 and got this:
  
[image]

The z_probe and z_min are separated and the z_probe is always triggered. Its confusing because the z_probe is using the same output for BL touch and fixed_mounter_probe and not anything to do with the sensorless homing Diag pin!!?!?. 
 Ive then checked out if I could use the unchecking the: 
[image]
   
but it sends an compiling error its not allowed. Okay. 

Then I tried unchecking the: 
[image]
And put in the PG8 pin witch is the Z_min endstop pin
This is also not an option an gets a compiling error.

So the solution for this mystery is to hardwire the z_probe with z_min endstop pin. This sould be okay because the PG8 pin (z_min) aka (Z-Diag pin) is 3.3volts, sending it to z_probe (PA2) witch can receive up to 5volts signal.

In my case on my skr pro 1.1 here with the green color: 
[image]

 Or here using my white wire in real life picture.
[image]

 
After hardwirering i send a M119 and get z_probe: open:
 [image]

Reflection:
This is a big confusion between two pinouts and why there should be two different pinouts I don’t know but believing that sensorless homing and probing sould be trigger by the same stallguard is my goal, you guys decide if this is a bug or calibration mistakes. I know what my answer is. Headless.

Perspective:
The only question I would ask is, why does those two pins separated

Conlcusion:
Z_probe open will now trigger proportional with z_min. that’s meaning Sensorless Homing and Sensorless Probing areboth now triggered by TMC Stallguard. Auto Bed leveling can now begin.
 
 Some pictures of the leveling done now:
[image]
[image]