MarlinFirmware/Marlin #24474: [FR] Independent simultaneous motion of several axes with individually set speed

mikhail11111, 2022-07-09:
### Describe the feature you want

It seems there is no way for Marlin to move several axes with individually set speeds simultaneously. i.e., the speeds are set so that to finish motion at the same time. 
Is it possible to have this feature?

sjasonsmith, 2022-07-09:
Synchronized movement is the primary point of Marlin. Right now to achieve what you want I believe your gcode generator would need to do the math to turn the single command you want into multiple synchronized moves with the correct contrived federate to make them seem disconnected.

You should probably consider whether they should actually be controlled by the same controller at all, if you don’t intend their motion to be coupled.

mikhail11111, 2022-07-10:
I was also thinking about the possibility to break the command into multiple moves, but that seems quite complicated. 
I think Marlin uses greater calculations to make the movement synchronized, and what I want should be actually easier (maybe a special case of synced movement). So shouldn't be difficult to implement programmatically. 
Multiple controllers is definitely the easiest solution, but then I would need e.g. 9 controllers instead of 1 (if a 9-axis Marlin version is used).

mikhail11111, 2022-07-15:
Basically, only the part of the code where the feedrates for each axis are calculated needs to be modified. 
Can somebody point to that part of code?

DerAndere1, 2022-07-16:
The speed of each axis is calculated in the planner: 
https://github.com/MarlinFirmware/Marlin/blob/8a1643587becba9a303572a451ee250de1865919/Marlin/src/module/planner.cpp#L2360-L2422

But RS274 G-code is designed for sequential moves of multiple axes. To deal with multiple feedrates, you would have to design a new command. So you also need to extend the parser. To get an Idea what has to change to add one additional "feedrate", see https://github.com/MarlinFirmware/Marlin/pull/24334/files .

For independent control of steppers, rather have a look at this library: https://github.com/luni64/TeensyStep .