MarlinFirmware/Marlin #24475: [BUG] Servo reverting position... Looks like i've found problem after 2 years.

BubonicPestilence, 2022-07-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm using DIY Opto Servo Probe (IR sensor), and pin is locked by servo.
I never really had any problems with it, except 1, which i though was solved by adding inductor around cables.
But it still happens... rarely... 1 in 30-40 deploys

Problem:
When marlin tries to deploy probe, in my case move servo to release pin position. Pin goes does, then servo goes back to 0 position. Looks like stutter, but it isn't, servo isn't stuttering.

Also tried solving it by using increased SERVO_DELAY. Didn't helped.

And now i probably understood problem.
If i **_disable_** `DEACTIVATE_SERVOS_AFTER_MOVE`, problem is gone.
My imagination makes me think: There is possibility when `DEACTIVATE_SERVOS_AFTER_MOVE` enabled and motors didn't finished moving (noise/inertia?), servo will be revered back by Servo::detach(), which will fail "waiting for probe-trigger" and leads to "probing failed" critical.

SKR 1.3, marlin-bugfix-latest

### Expected behavior

Servo unlock pin
Pin drops
Marlin probes bed
Servo goes up (to 0) and locks pin

### Actual behavior

Servo unlock pin
Pin drops
Servo goes up (to 0) and locks pin
Marlin fails probing

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Jun 22 2022 18:20:43) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Anet A8, upgrade to AM8

### Electronics

SKR 1.3, SKR 1.4

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

IdeaMaker

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

BubonicPestilence, 2022-07-09:
Possible places to find problem:

Servo::detach() shouldn't reset position to 0
Marlin triggers "move stop" somehow, and forcing servo::detach() (is it possible that marlin moves while waitinf for servo_delay?)

BubonicPestilence, 2022-07-09:
4 runs of `G29 E1` with 25 points.
0 fails.
Looks like i was right :)

sjasonsmith, 2022-07-10:
It’s possible that when the Servo is detached it ends up cutting the final pulse short or leaving the line high. Maybe those cause the misbehavior.

We recently fixed this for DUE-based boards, but it probably still could happen for some others.

closed 2022-09-19 by github-actions[bot]