MarlinFirmware/Marlin #24478: Change Filament or pause height

DocMarvel, 2022-07-10:
### Is your feature request related to a problem? Please describe.

When printing and needing a pause at layer or filament change code in Cura, the firmware pauses but does not give and option to resume the print. It doesnt bring up a notification. It pauses, homes, and stays. then it doesn't allow you to resume.

### Describe the feature you want

A feature that allows the use of post codes in cura to work.

oliver-eifler, 2022-07-13:
It's not marlin releated. the stock touch display can't handle the messages for notifications/dialogs

thisiskeithb, 2022-07-13:
Support for Anycubic's Vyper motherboard & TFT have not been added in this repo, so you'll need to make the request directly in the Vyper fork: https://github.com/ANYCUBIC-3D/Vyper/