MarlinFirmware/Marlin #24479: [BUG] ESP32 boards WEB UI Connection issue with ESP3DLib 1.2.0

3DSmitty, 2022-07-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

ESP3D looses web connection after approximately 30 seconds. This is caused by the "^" in ini/features.ini line 242 `luc-github/ESP32SSDP@^1.1.1.` The "^" causes Marlin to download and compile ESP32SSDP version 1.2.0 which is not compatible with Marlin's ESP3D. See:

https://github.com/luc-github/ESP3D/issues/784

https://github.com/luc-github/ESP3DLib/issues/44

### Bug Timeline

new bug started with 2.1

### Expected behavior

Not to loose web connection.

### Actual behavior

Looses web connection after 30 seconds.

### Steps to Reproduce

?

### Version of Marlin Firmware

2.1

### Printer model

Monoprice MP10

### Electronics

MKS Tinybee

### Add-ons

none

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Slic3r

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Bob-the-Kuhn, 2022-07-12:
Strange - BLAME says that` luc-github/ESP32SSDP@^1.1.1` has been there for at least 14 months.  

Most likely this has been an issue well before 2.1.

3DSmitty, 2022-07-12:
Yes could have been like that for a while if not forever. ESP32SSDP version 1.2.0 was released on March 24, 2022. That means anyone that compiled Marlin for esp32 after then would have this issue.

thisiskeithb, 2022-07-12:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/24489

closed 2022-07-12 by thisiskeithb

3DSmitty, 2022-07-12:
@thisiskeithb thanks!