MarlinFirmware/Marlin #24480: M149 ( set temperature unit ) not working

OneMonkeyArmy, 2022-07-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Of course, to use M149, TEMPERATURE_UNITS_SUPPORT must be define in Configuration.h

When using M149, it reports the current unit correctly..

But when trying to change to C K or F, using M149 C, M149 K or M149 F , it dosen't work, because the gcode parser does not see only a letter, it needs a value, as it is right now. If I use M149 and add a number after the UNIT Letter like: M149 F8 OR M149 K22, IT WORKS!!!

Of course the value has no effect. but it let accept the letter of the unit by the parser.

### Bug Timeline

I have no information on that.

### Expected behavior

An d before, someone think, I have mistaken the Display unit with the unit found in Gcode by the parser... I know it is not changing the display unit.

M149 C , should set the unit to Celcius.
M149 F , should set the unit to Farhenheit
M149 K , should set the unit to Kelvin.

But none of them does, unless you add a number after the Letter used as a flag for the desired unit

### Actual behavior

 M149 C67.. change to Celcius,
 M149 K1 .... works, M149 F777... you get the picture...
 
 It is probably not the only Gcode affected by this bug.

       
       

### Steps to Reproduce

1. VERYFY CURRENT UNIT BY ENTERING M149
   the answer is ->> M149 C ; Units in Celsius (OK IT IS THE DEFAULT!)
  
2. TRY TO CHANGE TO KELVIN
   ENTER M149 K, as described in the Gcode documentation form Marlin.org website
   the answer is ->> M149 C ; Units in Celsius 
   
3. VERYFY CURRENT UNIT BY ENTERING M149
  the answer is still->> M149 C ; Units in Celsius 

4. Bur if you enter M149 K6666
  the abswer is now->> OK !

5. VERYFY CURRENT UNIT BY ENTERING M149
  the answer is now -->M149 K ; Units in Kelvin
  
  
  

### Version of Marlin Firmware

2.0.9.3 / 2.094 / 2.1

### Printer model

tested on 3 different: Dremel 3D20, Flashforge Dreamer Nx, and custom build driven by BTT skr 1.4 turbo

### Electronics

n/a

### Add-ons

n/a

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Rear it, if you want to laugh...
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/9079380/Config.zip)

ellensp, 2022-07-10:
well  the fist thing I see in the code is different code depending on if you have FASTER_GCODE_PARSER enabled or not...

But since you didn't include your actual configs, "You do not need my config files," I cannot proceed

ellensp, 2022-07-10:
This was found and fixed a week ago.
You did not try bugfix [
](https://github.com/MarlinFirmware/Marlin/pull/24430)

closed 2022-07-10 by ellensp

OneMonkeyArmy, 2022-07-10:
It was well explained... You can recreate it in 5 seconds...  And because my marlin is so heavily modified. I asked my friend to check also, and he came back with the same result..Even if I send you my whole source, you will be completely lost...  All parts have been studied, modified, reduced, and optimized for my custom hardware, needs, and new possibilities, Planner, motion, stepper, thermal manager, are  very different. My GUI uses only my graphic library that have been written from the ground up and talk directly to the The TFT using DMA with FSMC to free my MCU...I will correct it myself... The parser is the last part I have to do... I kept it for the end.. It is the fun part... AND I AM SURE TO FIND MANY OTHER FUN THING IN IT !!!! But if you back off thinking you loose your time investigating my comments...because I didn't give all the requested information...

OneMonkeyArmy, 2022-07-21:
It is a real Bug, or should I call it an omission during some code updating... very easy to recreate, as explained... Why wasting my time trying to explain what I find here.. it goes nowhere... I have found another small thing to correct in the module  "temperature.cpp"... but why bother... nobody will take the time to check my discovery, because I do not include my configs... I will continue to correct them myself... and stop wasting my time here...