MarlinFirmware/Marlin #24481: [FR] Allow for composable configuration files

ucirello, 2022-07-10:
### Is your feature request related to a problem? Please describe.

One of the daunting aspects of the Marlin workflow is to go through both `Configuration.h` and `Configuration_adv.h` - probably a remnant of the Arduino IDE workflow.

I decided to investigate how bad the problem got over time, and after some digging at git history, I got these numbers:
```
$ git log --reverse --since='2018-01-01' --oneline Marlin/Configuration.h | while read commit _; do echo -n "$(git show -s --format='%as' $commit | awk -F '-' '{print $1"-"$2}');" ; git show $commit:Marlin/Configuration.h | wc -l | xargs; done | awk 'BEGIN{FS=OFS=";"} {if (a[$1]<$2) {a[$1]=$2; data[$1]=$0}} END{for (i in a) print data[i]}'  | sort
2017-11;1794
2018-01;1798
2018-02;1846
2018-03;1849
2018-04;1885
2018-05;1932
2018-06;2033
2018-07;1953
2018-08;1956
2018-09;1984
2018-10;2035
2018-11;2019
2018-12;2031
2019-01;2065
2019-02;2109
2019-03;2108
2019-04;2133
2019-05;2192
2019-06;2193
2019-07;2206
2019-08;2218
2019-09;2216
2019-10;2218
2019-11;2211
2019-12;2217
2020-01;2244
2020-02;2249
2020-03;2259
2020-04;2268
2020-05;2276
2020-06;2311
2020-07;2363
2020-08;2423
2020-09;2424
2020-10;2485
2020-11;2530
2020-12;2598
2021-01;2611
2021-02;2690
2021-03;2726
2021-04;2735
2021-05;2751
2021-06;2877
2021-07;2920
2021-08;2931
2021-09;2960
2021-10;2970
2021-11;2977
2021-12;2986
2022-01;2991
2022-02;2995
2022-03;2995
2022-04;3117
2022-05;3099
2022-06;3099
2022-07;3272
```

Essentially, `Configuration.h` grew 1800 raw lines of code to 3200. I believe a lot of that is improved documentation, but also a lot of that, I believe, can be accrued to increase in functionality. 

`Configuration_adv.h` has pretty much the same numbers (from 1900 in 2017 to 3200 in 2022: `git log --reverse --since='2018-01-01' --oneline Marlin/Configuration_adv.h | while read commit _; do echo -n "$(git show -s --format='%as' $commit | awk -F '-' '{print $1"-"$2}');" ; git show $commit:Marlin/Configuration.h | wc -l | xargs; done | awk 'BEGIN{FS=OFS=";"} {if (a[$1]<$2) {a[$1]=$2; data[$1]=$0}} END{for (i in a) print data[i]}'  | sort`)

C++11 offers conditional file inclusion (https://en.cppreference.com/w/cpp/preprocessor/conditional) and I wonder if we could use this technology to tackle the problem. 

Hence this proposal.

### Describe the feature you want

I would like to make a backward compatible change and use composable configuration files. 

I am not sure yet how the final shape looks like. I did an initial stab at https://github.com/MarlinFirmware/Marlin/commit/eee445ee20f16320432b2b5cd14665114212892a - it became quite evident how so many concepts got intertwined. 

Still, I believe it is possible to pluck theme based configuration files that can be drag'n'dropped into `Marlin/` that a user just opens and adjust all the things relevant to their use case.

It also means two important things: a) the very long generic files remain; and b) some kind of template based system might be necessary to extract blocks of codes into theme based files.

For debate purposes, let's consider:

```
Configuration_cartesian_additive.h
Configuration_corexy_additive.h
```

There would be a lot of overlap between these two files, and of course, we wouldn't want to have to update multiple files every time a feature is added or modified. And again for debate purposes, let's assume `Configuration.h` looked like

```
// ...

///@ template_group: PRINTER_VOLUME
// The size of the printable area
#define X_BED_SIZE 200
#define Y_BED_SIZE 200

// Travel limits (linear=mm, rotational=°) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 200

// ...
```

I would like to propose a parser (perhaps in Go - as my preferred language) that would scan the file for these `///@` template markers, and then use them for generation topic base configuration files. 

For example, `Configuration_cartesian_additive.h.tpl`:
```
{{ .GPLHeader }}
// ...
{{ .PRINTER_VOLUME }}
// ...
```

And, `Configuration_corexy_additive.h.tpl`:
```
{{ .GPLHeader }}
#define COREXY
// ...
{{ .PRINTER_VOLUME }}
// ...
```

InsanityAutomation, 2022-07-10:
This has been discussed at great length on discord and something along these lines will likely occur when enough of us can commit to a matching timeframe to skim all the example configurations in order to get them to the new format.

ucirello, 2022-07-11:
After some conversation with maintainers in `#dev-talk` it seems there are two orthogonal aspects uncovered by this issue.

One angle has to do with the technology used to create a configuration. Right now, Marlin uses two header files (`Configuration.h` and `Configuration_adv.h`). So there are some talks about potentially replacing them with something that reads better and perhaps more concise.

The other angle is how to manage the multitude of options in the current paradigm. 

I am going to retract this feature request for now while I think deeper about the what could constitute a good solution for the latter without contaminating the process with the concerns of the former.

closed 2022-07-11 by ucirello