MarlinFirmware/Marlin #24496: [BUG] Graphical LCD Artifacts in Marlin 2.1

AlexValCar, 2022-07-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Graphical LCD Artifacts are displayed on the screen with Ender 3 Max Creality 4.2.2 board within minutes of turning on the machine with firmware 2.1

### Bug Timeline

new bug, fine with Marlin 2.0.x

### Expected behavior

I expected that Graphical LCD Artifacts would not be displayed

### Actual behavior

Graphical LCD Artifacts were displayed minutes after turning on the machine and entering menus

### Steps to Reproduce

1. turn on the printer
2. enter the manual bed leveling option

### Version of Marlin Firmware

2.1

### Printer model

Ender 3 Max

### Electronics

Stock 4.2.2 board

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.h and Configuration_adv.h.zip](https://github.com/MarlinFirmware/Marlin/files/9108471/Configuration.h.and.Configuration_adv.h.zip)

hcet14, 2022-07-14:
Have you had a look here ['https://marlinfw.org/docs/basics/troubleshooting.html'](https://marlinfw.org/docs/basics/troubleshooting.html)?

AlexValCar, 2022-07-14:
Yes, unfortunately it is not updated and the parameters it shows are no longer in the firmware and that is exactly where I got the title of the issue