MarlinFirmware/Marlin #24498: Get current position with time

pawelrekas, 2022-07-14:
### Is your feature request related to a problem? Please describe.

I would like to synchronize the printer with another device in such a way that the position of the carriage and the moment of time when this position was known.

I came up with the following way:
1. set some time on the controller to make it common with the other device.
2. retrieve carriage position data over time, e.g. in the form: X, Y, Z, TIME.

Is it possible to implement such functionality with MARLINE software?

Translated with www.DeepL.com/Translator (free version)

### Are you looking for hardware support?

Arduino MEGA + REPRAP 1.4 + 3x 2HSS57, Delta printer.

### Describe the feature you want

Functionality involving the ability to synchronize the printer with another device so that the other device knows at what moment in time and in what position the carriage is.

ellensp, 2022-07-18:
https://marlinfw.org/docs/gcode/M154.html
Marlin has no time, no real time clock, no manual clock.  It would be up to the receiving program to add a time stamp.

closed 2022-07-18 by ellensp

pawelrekas, 2022-08-04:
Thank you for your answer.
I read about the M154 command before, but I wonder if it will be a significant delay between the moment when Arduino, aka Marlin, will send data with position and the moment when the computer receives it (signal transmission path is: 1) send data with Arduino, 2) convert data from RS to USB via CH340 and 3) receive data in USB at the computer). Anyway, I know there is no option to add RTC now to the system at the Arduino/Marlin level.