MarlinFirmware/Marlin #24503: [BUG] butloader does not works

gromas, 2022-07-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi. I have sanguino based 3dprinter and upload marlin sketch with bootloader to 1284p. Sketch works but bootloader does not works correctly.
I try to upload sketch with USB cable and get error 
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00

Then I lock for output files and ivestigate size of hex with bootloader lower than size hex without bootloader. What it is?
[image]

### Bug Timeline

new bug but I dont u

### Expected behavior

Create sketch for sanguino and export hex's. Check file sizes.

### Actual behavior

Image size with bootloader lower than image size without.

### Steps to Reproduce

Create sketch for sanguino 
Export binaries. 
Check file sizes.

### Version of Marlin Firmware

Marlin bugfix-2.1.x

### Printer model

Prusa I3

### Electronics

Melzi 2.0

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Bob-the-Kuhn, 2022-07-14:
I don't know if it is causing the AVRDUDE errors but it looks like the image (153K) is bigger than the available FLASH in the controller (about 120K).  You'll need to delete/disable options until your image is under 120K and try again.

thisiskeithb, 2022-07-14:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-07-14 by thisiskeithb

ellensp, 2022-07-14:
@Bob-the-Kuhn thats a intel hex file, it is a text representation of the bin. you just convert it back to binary to get its real size.
@gromas  with bootloader means that the image contains a bootloader, you need to use a AVRISP to program the device. As that is the only way you can write a new bootloader.