MarlinFirmware/Marlin #24504: [BUG] Steps/mm halved after update from 2.0.X

Woutah, 2022-07-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have recently updated my firmware from version 2.0.X to the most recent 2.1.X, but the steps per millimeter for both the Z-axis and extruder differ by a factor of 2. 
On version 2.0.X (2021.07) I have the following settings:
#define DEFAULT_AXIS_STEPS_PER_UNIT    { 80, 80, 800, 138.9 }
 
To get accurate prints in version 2.1.X, I have to edit them to:
#define DEFAULT_AXIS_STEPS_PER_UNIT    { 80, 80, 400, 69.5 }

I rolled back and forth several times to make sure, I kept pretty much all non-default settings the same between versions. 
I attached the config files for both builds. 

NOTE: My printer does give a "TMC CONNECTION ERROR" when starting up (although it has always printed fine), this may be related?

### Expected behavior

No change in steps per unit when changing versions. 

### Actual behavior

Steps per unit changes. 

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Jul 15 2022 19:21:25) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Creality Ender 5

### Electronics

BOARD_BTT_SKR_V2_0_REV_B, BLTOUCH, TMC2209

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/9122977/Configurations.zip)

ellensp, 2022-07-16:
Steps/mm is set for a specific micro stepping level.

Micro stepping is set via uart to the stepper driver which is not working  "TMC CONNECTION ERROR"

Power on microstepping settings are not controlled by marlin.

Fix your TMC ERROR

closed 2022-07-16 by ellensp

Woutah, 2022-07-16:
Thanks for your input. Isn't it weird then that the behaviour differs between the two versions?
With my limited knowledge I would assume both versions to have the same issues if the problem lies in communication to/from the drivers.