MarlinFirmware/Marlin #24505: auto bed leveling error

Pieter005, 2022-07-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

hi guy's 

when i sent the g29 commant to the printer for auto bed leveling it starts leveling it starts good bud the movement to the left goes wrong it moves a litle bit and starts probing. after doing this 2 times it moves to the right in the normal distance. eventualy it moves out the bed and i have to send m112 to stop it. i tried '''#define PROBE_Y_FIRST''' but then it won't move to the front.
and '''Z_STEPPER_AUTO_ALIGN'''have the same issue. but when printing it is not

printer: hypercube evolution
mainboard: btt octopus v1.1 stm32f466
stepper driver: tmc2209
kinematics: core XY
z axis: dual leadscrew
display: no display 
microsteps: 256 (also tried 16)

------------------
:           7        8   :     9
:                          :
:           6    5   4  :
:                          :
:  1        2        3  :
------------------

### Bug Timeline

it is a new build i tried marlin 2.0 and 2.1

### Expected behavior

i expected to do its normal movement and do its bed leveling how it should be.

### Actual behavior

it dont move how it should be

### Steps to Reproduce

1. i send the ''g28'' command so it can home.
2. i send the ''g29'' command for auto bed leveling

### Version of Marlin Firmware

''Marlin-bugfix-2.1''  and i also tried ''Marlin-2.0.9.4''

### Printer model

hypercube evolution dual z motor

### Electronics

diy system btt octopus

### Add-ons

direct drive voron afterburner

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configuration hypercube.zip](https://github.com/MarlinFirmware/Marlin/files/9125473/configuration.hypercube.zip)

Bob-the-Kuhn, 2022-07-16:
First - a couple of assumptions:
- G28 (homing) is working correctly.  The text "now it doesn't move to the left or front during homing" is referring only to the G29 command.
- You **had been** using a standard version of Marlin (2.0.8.0, etc, not a manufacturer's customized version) and only had problems after updating to 2.0.9.4 and/or Marlin-bugfix-2.1.  We've seen update issues where the mfgr had made changes to files besides the config files.

Most of the "out of bounds" probing issues can be traced to one of the following.  Please double check them.
```
NOZZLE_TO_PROBE_OFFSET
X_BED_SIZE
Y_BED_SIZE
X_MIN_POS 
Y_MIN_POS 
```

I'm worried about NOZZLE_TO_PROBE_OFFSET.  A { 10, 10, 0 } offset seems rather small for most Z probes.  Definitely too small if using a BLTouch type of probe.

thisiskeithb, 2022-07-16:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-07-16 by thisiskeithb

Pieter005, 2022-07-16:
hi Bob thankyou for the reply i fixed the probe offset but that didn't work.X_BED_SIZE Y_BED_SIZE X_MIN_POS Y_MIN_POSthe ''bed_size" and "min_pos'' are set to the richt value to.I am buiding this printer new so it is not due to a firmware update. it may be an editting mistake I started with marlin 2.1 and becouse of the error i tried 2.0. it can't be a mechanical or electrical issue becouse it only happens when running the ''g29'' or ''g34''. I marked the spots of the 3x3 mesh on the bed so u can see what i mean.(see picture) after point 7 it moves out of range and i have to stop it with ''m112''when i print without bed leveling it prints finekind regards, Pieter

thisiskeithb, 2022-07-16:
This isn’t a support forum, so please use the links in [my comment above](https://github.com/MarlinFirmware/Marlin/issues/24505#issuecomment-1186239876) for configuration issues.