MarlinFirmware/Marlin #24515: [BUG] X_DUAL_ENDSTOPS enabled and X_HOME_DIR +1 fails to compile

aughey, 2022-07-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to build the V1 Lowrider 3 firmware in a configuration with X_DUAL_ENDSTOPS enabled and X_HOME_DIR set to "+1".  This is a recurrence of https://github.com/MarlinFirmware/Marlin/issues/22484  The commit applied to this issue was inadvertently reverted by a subsequent commit - at least partially.

I've attempted to make the changes made back in August without success.  Changing the relevant sections in endstops.h and Conditionals_post.h now fails at

```
In file included from Marlin/src/module/../inc/../HAL/./AVR/HAL.h:26:0,
                 from Marlin/src/module/../inc/../HAL/HAL.h:30,
                 from Marlin/src/module/../inc/MarlinConfig.h:31,
                 from Marlin/src/module/endstops.h:28,
                 from Marlin/src/module/endstops.cpp:27:
Marlin/src/module/endstops.cpp: In static member function 'static void Endstops::update()':
Marlin/src/module/endstops.cpp:945:65: error: 'X_MAX' was not declared in this scope
              if (stepper.axis_is_moving(X_AXIS)) { _ENDSTOP_HIT(X, TERN(X_HOME_TO_MIN, MIN, MAX)); planner.endstop_triggered(X_AXIS); }
                                                                 ^

```

I've tried this with bugfix-2.0.x branch and the 2.1.x branch without success.  

### Bug Timeline

new bug with new configuration

### Expected behavior

Compile

### Actual behavior

fails

### Steps to Reproduce

opt_enable X_DUAL_ENDSTOPS
opt_set X_HOME_DIR "+1"

### Version of Marlin Firmware

2.0.x and tried 2.1.x

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9128763/Configuration.zip)

Bob-the-Kuhn, 2022-07-18:
Try changing, in configuration_adv.h the line `#define X2_USE_ENDSTOP    _XMAX_` to `#define X2_USE_ENDSTOP    _XMIN_`

When you tell Marlin to home to X_MAX it assumes that the XMAX endstop will be used.  `#define X2_USE_ENDSTOP    _XMAX_`  tellls Marlin that the second endstop to use is also XMAX.  That confuses Marlin and throws an error.

ellensp, 2022-07-18:
Config has 

#define X_HOME_DIR +1 // -1
#define Z_HOME_DIR +1 // -1
#define X2_DRIVER_TYPE   A4988
#define Z2_DRIVER_TYPE   A4988
with   #define X_DUAL_ENDSTOPS             // X2 has its own endstop
and  #define Z_MULTI_ENDSTOPS            // Other Z axes have their own endstops
 
but you have

    #define X2_USE_ENDSTOP    _XMAX_  // X2 endstop board plug. Don't forget to enable USE_*_PLUG.
    #define Z2_USE_ENDSTOP   _XMAX_   // Z2 endstop board plug. Don't forget to enable USE_*_PLUG.

X-Max and Z-max endstops are already in use due to homing to max on both x any z axis. So these cannot be used here

instead use

    #define X2_USE_ENDSTOP    _XMIN_  // X2 endstop board plug. Don't forget to enable USE_*_PLUG.
    #define Z2_USE_ENDSTOP   _ZMIN_   // Z2 endstop board plug. Don't forget to enable USE_*_PLUG.

This fixes all endstop issues.

There is how ever a bug in M350 gcode where by it is expecting E to exist
apply this diff

```DIFF
diff --git a/Marlin/src/gcode/control/M350_M351.cpp b/Marlin/src/gcode/control/M350_M351.cpp
index d0ab75bb00..e9088c272f 100644
--- a/Marlin/src/gcode/control/M350_M351.cpp
+++ b/Marlin/src/gcode/control/M350_M351.cpp
@@ -35,7 +35,9 @@
 void GcodeSuite::M350() {
   if (parser.seen('S')) LOOP_DISTINCT_AXES(i) stepper.microstep_mode(i, parser.value_byte());
   LOOP_LOGICAL_AXES(i) if (parser.seenval(AXIS_CHAR(i))) stepper.microstep_mode(i, parser.value_byte());
-  if (parser.seenval('B')) stepper.microstep_mode(E_AXIS + 1, parser.value_byte());
+  #if DISTINCT_AXES > 4
+    if (parser.seenval('B')) stepper.microstep_mode(E_AXIS + 1, parser.value_byte());
+  #endif
   stepper.microstep_readings();
 }

```

closed 2022-07-18 by ellensp