MarlinFirmware/Marlin #24516: [BUG] TMC 2130 does not work in SPI mode

dronecz, 2022-07-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I designed custom board for [FluidNC](https://github.com/bdring/FluidNC), but find out that there is not support (and it is not planned) for features which I need for my pick and place machine so I decide to switch to Marlin. 

My board is using 74HC595 shift registers to control stepper motors, but issue which I´m facing now is that I can not establish connection via SPI. This is defintely SW issue as if I upload FluidNC firmware, all axis are working fine. 

Here is my log from Printerface:

```
Connecting...
Attempted to write invalid text to console, which could be due to an invalid baudrate
 External Reset
 Brown out Reset
 Watchdog Reset
Marlin bugfix-2.1.x
echo: Last Updated: 2022-07-17 | Author: (none, default config)
echo: Compiled: Jul 18 2022
echo: Free Memory: 313444  PlannerBufferBytes: 1408
echo:Hardcoded Default Settings Loaded
echo:; Linear Units:
echo:  G21 ; (mm)
echo:; Temperature Units:
echo:  M149 C ; Units in Celsius
echo:; Steps per unit:
echo:  M92 X40.00 Y40.00 Z40.00 A4.44 B4.44 C40.00
echo:; Max feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z250.00 A50000.00 B50000.00 C5000.00
echo:; Max Acceleration (units/s2):
echo:  M201 X1000.00 Y1000.00 Z2000.00 A3000 B3000 C3000
echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:  M204 P1000.00 R3000.00 T3000.00
echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00 A0.00 B0.00 C0.00
echo:; Stepper driver current:
echo:  M906 X800 Y800 Z800 A700 B700 C800
echo:; Driver stepping mode:
echo:  M569 S1 X Y Z A B C
echo:  M569 S1  A
echo:  M569 S1  B
echo:  M569 S1  C
Testing X connection... Error: All HIGH
Testing Y connection... Error: All HIGH
Testing Z connection... Error: All HIGH
Testing I connection... Error: All HIGH
Testing J connection... Error: All HIGH
Testing K connection... Error: All HIGH
Printer is now online.
```
I searched issues here, searched Discord channels (looked for "I2S", "All LOW", "All HIGH"), but did not find solution. 

I tried use TMC_USE_SW_SPI, tried put X CS pin on real HW pin, but without change. I´m using BIGTREETECH TMC2130 V3.0 .

As I´m using custom board definition, I´m including files so you can compile same code as me. 
[marlin_configs.zip](https://github.com/MarlinFirmware/Marlin/files/9128926/marlin_configs.zip)

I ordered ESP PROG so I will be hopefully able to use debugger and see what FW is doing. 

Thanks for help or any direction. 

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Jul 18 2022 02:26:39)

### Printer model

none

### Electronics

custom board

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Bob-the-Kuhn, 2022-07-18:
You may end up using the TMC2130s in standalone mode.  Is there a SPI only feature you need from these chips?

TMC_USE_SW_SPI assumes all SPI interfaces have a common TMC_SW_MOSI, TMC_SW_MISO & TMC_SW_SCK with dedicated chip selects.  I see the dedicated chip selects but not the others.  

I'm not aware that Marlin has the ability to control/use the 74HC595 shift registers.

I'm thinking this request is outside the scope of this forum.  This is dedicated to Marlin bug fixes.

There are a couple of Facebook Marlin support goups ([3Dtechtalk](https://www.facebook.com/groups/3Dtechtalk/) is one) plus a [RepRap one](https://reprap.org/forum/list.php?415) that are better suited to this type of question.

EvilGremlin, 2022-07-18:
Which SPI actually used for TMC? If you try using one shared with flash i wouldn't expect it to work at all.

ellensp, 2022-07-18:
@Bob-the-Kuhn esp32 has built into the HAL shit register support, to extend the number of IO pins. 
I dont know if you could software spi over it though.,,,

Bob-the-Kuhn, 2022-07-18:
I'm amazed at what they've been able to do with the ESP32 platform.  Definitely a world I haven't explored.

EvilGremlin, 2022-07-18:
CS on extended pins should be possibe anyway, with hw or sw SPI

dronecz, 2022-07-18:
> Which SPI actually used for TMC? If you try using one shared with flash i wouldn't expect it to work at all. 
You mean internal SPI flash which is in ESP32 module? That one is on SPI1 (AFAIK) and I´m using VSPI which should be SPI3 (pin 18,19,23). 

I understand that there is probably limitation in using shift registers in Marlin with current code, but if you support SPI drivers and settings current via SPI, I should be (at least) able to communicate with driver. 
As I meantioned, I tried initialize driver on X axis with GPIO CS pin (33), but result is the same as with shift registers. If I´m not mistaken, I supplied all necessary pins to communicate over SPI (SDI, SDO, CLK, CS).

> With the current esp-idf, you could simulate it by adding your devices as one device (without a CS pin) to the spi driver, then defining a pre- and post-transaction callback that pulls down the GPIO connected to the CS of the device you actually want to control. 

ESP-IDF change in last 4 years so it is possible to control CS pin via I2S and shift registers. [FluidNC](https://github.com/bdring/FluidNC/blob/main/FluidNC/src/I2SOut.cpp) has code for that. 

I would like to help to resolve this, but I would need some point where to start as I do not know how Marlin internally works.

stef-ladefense, 2022-09-07:
* To use TMC2130, TMC2160, TMC2660, TMC5130, TMC5160 stepper drivers in SPI mode
 * connect your SPI pins to the hardware SPI interface on your board and define
 * the required CS pins in your `pins_MYBOARD.h` file. (e.g., RAMPS 1.4 uses AUX3
 * pins `X_CS_PIN 53`, `Y_CS_PIN 49`, etc.).
 * You may also use software SPI if you wish to use general purpose IO pins.

EvilGremlin, 2022-09-07:
LOL, this much is obvious. Please read about esp-idf SPI peculiarities.

dronecz, 2022-10-31:
I finally figured out what was my problem. After a lot of testing I found out that I can not toggle CS pin at TMC driver if this pin is driven with shift register and I2S. When I control just STEP and DIR pin via I2S and use some GPIO for CS pin, it works as expected. Does anyone know code for I2S? 

I will try to make PR for documentation with my findings as it will definetely help other people.

EvilGremlin, 2022-11-01:
esp-idf framework is not aware about I2S pins, you'll have to redefine/overload any SPI CS related functions in HAL to use them

closed 2023-01-11 by github-actions[bot]