MarlinFirmware/Marlin #24517: [BUG] Robin Nano V3 with BTT MAX31865 V2.0 reads 989°C, MINTEMP:E1

trashypopashy, 2022-07-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When booting the printer, the display first shows a hotend-temperature of 989 °C, then warns "Err: MINTEMP: E1 PRINTER HALTED Please Reset"
If I don't set the max-temp for the hotend high enough (say I leave it at 275 °C, which is default for Marlin), the printer states: "Err: MAXTEMP: E1 PRINTER HALTED Please Reset" at the very beginning (I can't see the homescreen with temperatures etc. at all).

### Bug Timeline

new, startet when i tried to add the MAX31865

### Expected behavior

I expect to get the correct temperature with a combonation of MKS Robin Nano V3, BTT MAX31865 V2.0 an a PT100.

### Actual behavior

Temperature reads 989 °C, after some seconds the display states MINTEMP:E1

### Steps to Reproduce

- add the files included in the zip-file to the latest marlin: [archive.zip](https://github.com/MarlinFirmware/Marlin/files/9135527/archive.zip) - just these 3 files are edited by me, to keep the reproduciont of this problem as simple as possible
- compile firmware
- add BTT MAX31865 V2.0 an PT100 to E1-stepper-slot to a MKS Robin Nano V3
- flash firmware
- boot Printer

### Version of Marlin Firmware

DISTRIBUTION_DATE "2022-07-18

### Printer model

Sapphire Plus

### Electronics

MKS Robin Nano V3

### Add-ons

BTT MAX31865 V2.0

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-07-18:
Duplicate of #24513

closed 2022-07-18 by thisiskeithb

trashypopashy, 2022-07-18:
Sorry for asking on in this Bug-report: Could it be possible, I ddin't configure SPI in Marlin correctly?

trashypopashy, 2022-07-28:
Today I found out what was the problem in my case: I inverted the DIP-switches in the MAX31865-module. Now it is the way, Bigtreetech wrote it in the instructions. Sorry for the confussion.