MarlinFirmware/Marlin #24522: [FR] (feature summary)Switch or toggle a pin at M600 Pause

Deltaflyer7, 2022-07-19:
### Describe the feature you want

I would like to switch a user definable pin when a break by M600 (Filament Change via GCode, Filament Runout). The pin should either be reset after the resume, or simply output a pulse of e.g. 50ms. Purpose of the action: Possibility of signaling the user request in another room, for printers that do not use WiFi, Octoprint or similar, using a visual alarm (e.g. for the hearing impaired).

Deltaflyer7, 2022-07-19:
Thanks very much. Does this modification also ensure that the pin is reset either after e.g. 50ms or after the resume, or where exactly do I have to use the 2nd M42 so that the pin is switched off again?

Deltaflyer7, 2022-07-19:
Sorry, how do I update the filament runout script and where can I find this script? I use Marlin 2.1

thisiskeithb, 2022-07-19:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Deltaflyer7, 2022-07-19:
OK thank you.I have now turned to one of the recommended forums.