MarlinFirmware/Marlin #24526: [BUG] LASER_FEATURE incompatible with EXTRUDERS 0

040medien, 2022-07-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If you set `EXTRUDERS 0` and enable `LASER_FEATURE`, building the firmware fails with this error:

`Marlin/src/module/temperature.cpp: In static member function 'static void Temperature::isr()':
Marlin/src/module/temperature.cpp:3264:56: error: 'gcode' was not declared in this scope
     if (cutter.last_power_applied && ELAPSED(millis(), gcode.previous_move_ms + (LASER_SAFETY_TIMEOUT_M
S))) {
                                                        ^
 `

### Bug Timeline

Since Marlin-2.1.0

### Expected behavior

You can compile with both 

### Actual behavior

Compilation fails

### Steps to Reproduce

1. set EXTRUDERS 0 and enable LASER_FEATURE
2. remove extruder related values for DEFAULT_AXIS_STEPS_PER_UNIT, DEFAULT_MAX_FEEDRATE, DEFAULT_MAX_ACCELERATION, AXIS_RELATIVE_MODES
3. build

### Version of Marlin Firmware

Marlin 2.1 (Jul 20 2022 18:20:45) 

### Printer model

LowRider CNC V3

### Electronics

SKR PRO v1.2

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/9152040/Configs.zip)

EvilGremlin, 2022-07-20:
you forgot to disable TEMP_SENSOR

040medien, 2022-07-20:
@EvilGremlin Actually in my actual config I have disabled TEMP_SENSOR_0 and it doesn't make a difference. Apparently there is some logic regarding laser safety timeouts contained in temperature.cpp - so this gets included even if you don't have any temperature sensors. Some separation of concerns might be helpful here - but that's goes beyond my meagre abilities.

Given that CNC setups with no extruders are a bit more exotic this is not the first regression that goes unnoticed for a while. It would be great to have a test with such a setup that enables some CNC and spindle or laser features in the settings, so this would be caught in the CI pipeline. @thinkyhead wdyt?

EvilGremlin, 2022-07-20:
ah, got it. temperature is a little nightmare now, Scott planned to refactor it into someting modular and coherent

thisiskeithb, 2022-07-20:
Closing since https://github.com/MarlinFirmware/Marlin/pull/24527 has been merged.