MarlinFirmware/Marlin #24529: [BUG] So if I have the GD32F303 chip am I screwed out of using BLTouch?

jmeeter, 2022-07-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Bought a Ender 3 V2 and immediately my friend told me to buy a BLTouch. I did so and now after 6 hours of Googling, am realizing the issue is that I have the GD32F303 RET6 CPU. After talking to a guy in a FB group he said if I upgraded my screen to the Pro's touch screen then there's custom firmware I can run. Then he pointed me to JYers which seems to be popular.

Anyway, the tl;dr is - if I have a Ender 3 V2 with the 4.2.2 board and GD32F303 RET6 chip, what are my options for BLTouch? Do I shelve the BLTouch and wait for Creality to release new firmware or return the BLTouch?

### Version of Marlin Firmware

1.0.4

### Printer model

Creality Ender 3 V2

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-07-21:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/21473

thisiskeithb, 2022-07-21:
GD32-based boards are not officially supported, but users have had luck using `*_maple` environments.

closed 2022-07-21 by thisiskeithb