MarlinFirmware/Marlin #24530: Delta Probe XY offset installation then ABL problem

PandaBalls, 2022-07-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

1. The Images show how I installed the Probe Sensor on my Delta Printer. It +Y 22mm offset..
[image]
[image]

2. After Delta Auto Calibration and UBL, the Bed Always tiled in one direction. As Photo below.
[image]

3.I tried use the Nozzle Pressure Sensor or Installed the sensor just under Nozzle. ABL works fine.
[image]
[image]

### Expected behavior

I expected Marlin will take care of the Probe offset in UBL.

### Version of Marlin Firmware

2.1

### Printer model

LX Delta

### Electronics

MKS gen 1.2

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/9156729/Configurations.zip)
[ABL logging.txt](https://github.com/MarlinFirmware/Marlin/files/9157017/ABL.logging.txt)

Bob-the-Kuhn, 2022-07-21:
I take it this is a piezo sensor.  Feedrate affects when it triggers and how repeatable the triggers are.  I suggest setting Z_PROBE_FEEDRATE_FAST and Z_PROBE_FEEDRATE_SLOW the same and experimenting.

Please include the results of the M420 V3 command next time so we can see the full matrix.

Is it too low all along the back edge or just in the one spot closest to the edge?

PandaBalls, 2022-07-22:
> 
The probe repeatable M48 Test Result is: 0.001, it works well. And the Probe Slow Feedrate is 100, I think it is slow enough.

Here are the Two UBL Matrix: 

01. Problem One ( Probe installed at +X 22.5mm offset as the photo I posted) 
Bed Topography Report:
    (  -90, 90)                                      ( 90, 90)
        0       1       2       3       4       5       6
 6 | +4.29  +4.23  +4.17  +4.05  +4.08  +4.06  +4.06
   |
 5 | +4.28  +4.22  +4.17  +4.10  +4.08  +4.06  +4.06
   |
 4 | +4.21  +4.21  +4.24  +4.20  +4.20  +4.16  +4.16
   |
 3 | +4.31  +4.25  +4.28  +4.28 [+4.30] +4.29  +4.31
   |
 2 | +4.45  +4.41  +4.37  +4.43  +4.41  +4.39  +4.39
   |
 1 | +4.75  +4.68  +4.60  +4.59  +4.52  +4.51  +4.51
   |
 0 | +5.06  +4.95  +4.83  +4.79  +4.64  +4.63  +4.63
        0       1       2       3       4       5       6
    (  -90,  -90)                                      ( 90,  -90)
Mesh is valid
Storage slot: 1
----------------------------------------------------------------

02.This is the probe just under Nozzle(As the Photo I posted) , AND The UBL **works fine**:

Bed Topography Report:
    (  -90, 90)                                      ( 90, 90)
        0       1       2       3       4       5       6
 6 | +0.17  +0.04  -0.09  -0.14  -0.15  -0.13  -0.10
   |
 5 | +0.03  -0.03  -0.10  -0.14  -0.15  -0.13  -0.10
   |
 4 | -0.11  -0.11  -0.10  -0.12  -0.11  -0.08  -0.06
   |
 3 | -0.15  -0.16  -0.13  -0.18 [-0.10] -0.11  -0.06
   |
 2 | -0.16  -0.16  -0.15  -0.14  -0.14  -0.15  -0.15
   |
 1 | -0.01  -0.08  -0.16  -0.13  -0.16  -0.18  -0.18
   |
 0 | +0.15  -0.00  -0.16  -0.07  -0.16  -0.18  -0.18
        0       1       2       3       4       5       6
    (  -90,  -90)                                      ( 90,  -90)
Mesh is valid
Storage slot: 0

Bob-the-Kuhn, 2022-07-22:
I don't want to sound flippant but - if you have a matrix that works well then use it.  It usually takes me 3 or 4 iterations through G26 and G29 P4 before my mesh is usable.

PandaBalls, 2022-07-22:
> I don't want to sound flippant but - if you have a matrix that works well then use it. It usually takes me 3 or 4 iterations through G26 and G29 P4 before my mesh is usable.

The probe Under Nozzle , I need to install and uninstall it every time doing UBL.  But it did work well even 1 iteration.

But Install Probe +X 22mm and +Z 2mm , it is the best way to fix the probe there, It always standby there.. 
Unfortunately I can't make it work properly , unless I manually edit all Points..

tombrazier, 2022-08-20:
Could this be caused by some kind of delta analogue of X axis twist? e.g. maybe the print head's vertical axis varies with X/Y position. Then the 22mm X offset results in a Z offset that is dependent on the varying axis orientation.

closed 2022-10-31 by github-actions[bot]