MarlinFirmware/Marlin #24532: [BUG] Z-home does not work on TronXY X5SA board model CXY-V6-191017

zhanhongliau, 2022-07-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Z Homing doesn't do anything when I select the Auto Home option. X and Y work well, the printhead moves to the X and Y endstops, but when it returns to the center of the bed to do Z homing, the bed just drops down about 10mm, pauses for a split second, then moves downwards again another 10mm.

I've noticed that the Z home will work if I move the bed close enough to the Z probe that it activates (the light turns on). But this means that after every print where the bed ends up far from the probe (Z axis is way positive), if I turn off the machine I have to manually move the bed up towards the probe, then select autohome, and then hope the bed is properly aligned when I start printing.

It appears the board I have is the new 'silent motherboard' model CXY-V6-191017 with TMC drivers but I neglected to take a pic of it when I had the case opened.

I found an [older issue here](https://github.com/MarlinFirmware/Marlin/issues/23677) where a user toggled pullup on pin PA14 but I tried putting his code in the setup function in MarlinCore.cpp and that didn't seem to do anything. Building on chitu_f103 or chitu_v5_gpio_init doesn't seem to change anything either.

### Bug Timeline

Started when I updated the firmware from the stock version to a Marlin build

### Expected behavior

I expected Z home to work. The original firmware by TronXY I think is based on Marlin as well and I remember the two prints I did with it seemed to properly move the bed upwards to find Z home regardless of the starting position of the bed.

### Actual behavior

When it starts to do Z homing, the bed moves down a short distance, pauses, then moves down a short distance again and stops.

### Steps to Reproduce

1. Power off the printer
2. Move the bed some distance from the probe/nozzle, maybe about 30mm
3. Turn on the printer, then select Auto Home

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Jul 21 2022 09:44:52) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Tronxy X5SA V6 330 TMC EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

TronXY X5SA (330x330 bed size, non-pro)

### Electronics

Stock electronics. Board model CXY-V6-191017

### Add-ons

No changes to stock hardware

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/9160035/Marlin.zip)

Bob-the-Kuhn, 2022-07-21:
Enabling **#define ENDSTOPPULLUPS** should take care of your endstop pullup concerns.

Does M119 report the Z_MIN_PROBE as triggered when not at the home position?  If yes then invert **Z_MIN_PROBE_ENDSTOP_INVERTING** and **Z_MIN_ENDSTOP_INVERTING**.

zhanhongliau, 2022-07-22:
`Z_MIN_PROBE_ENDSTOP_INVERTING` and `Z_MIN_ENDSTOP_INVERTING` are already inverted (set to true), and M119 reports `Z_MIN_PROBE` as open when the bed is far from the probe so that seems fine. I tried to enable `ENDSTOPPULLUPS` and that ended up causing the limit switches for X and Y to stop working, causing the printer to try to move the head way outside limits (oops). Enabling specifically `ENDSTOPPULLUP_ZMIN_PROBE` doesn't do anything either, the bed still moves down a bit, pauses, then moves down even more and stops.

I also tried setting `Z_MIN_ENDSTOP_INVERTING` to false and that actually caused the bed to move upwards during the Z home instead of down! But unfortunately the probe does not trigger and it causes the nozzle to crash into the bed. 

With `ENDSTOPPULLUPS` disabled, and `Z_MIN_PROBE_ENDSTOP_INVERTING` set to false, the Z home completely doesn't work, even if the bed starts close to the nozzle the bed just shifts downwards a few times and stops. At least with `Z_MIN_PROBE_ENDSTOP_INVERTING` set to true it was able to home Z if the bed was manually placed close to the nozzle at the start.

Is there anything else I can do to debug this issue?

zhanhongliau, 2022-07-22:
Digging into this more it looks like M119 reports Z_min as TRIGGERED on startup! It seems that the key is to trigger the probe once after startup to have it properly detect its real state, only then will Z homing work. This can be as simple as waving my hand in front of the probe. 

I've tried building in the `chitu_v5_gpio_init` environment which supposedly has a bugfix for the incorrect probe state on startup, but that didn't seem to change anything. In `boards_setup.cpp` 
```
            /**
             * PA14 is a pull up pin. But, some V5 boards it start with LOW state! And just behave properly when the Z- PROBE is actived at least once.
             * So, if the sensor isnt actived, the PA14 pin will be forever in LOW state, telling Marlin the probe IS ALWAYS ACTIVE, that isnt the case!
             * Chitu original firmware seems to start with every pullup PIN with HIGH to workaround this.
             * So we are doing the same here.
             * This hack only works if applied *before* the GPIO Init, it's the reason I did it here.
             */
            #ifdef CHITU_V5_Z_MIN_BUGFIX
              GPIOA->regs->BSRR = (1U << PA14);
            #endif
```

Putting the pullup line of code in the setup function in MarlinCore.cpp gives me a compile error saying GPIOA has no such member regs. How would I go about forcing the probe to trigger itself once on startup?

zhanhongliau, 2022-07-22:
Wow, okay, I just tried adding
```
  
  SET_INPUT_PULLDOWN(Z_MIN_PIN);
  SET_INPUT_PULLUP(Z_MIN_PIN);
  
```

To the setup function and that fixed it. Z min does not show as Triggered on startup anymore, and Z home works properly regardless of the starting bed position.

Bob-the-Kuhn, 2022-07-22:
Fantastic!

Barrnet, 2022-08-08:
> Wow, okay, I just tried adding
> 
> ```
>   
>   SET_INPUT_PULLDOWN(Z_MIN_PIN);
>   SET_INPUT_PULLUP(Z_MIN_PIN);
>   
> ```
> 
> To the setup function and that fixed it. Z min does not show as Triggered on startup anymore, and Z home works properly regardless of the starting bed position.

I have this problem but I don't understand where put this lines on the file. Could you provide an example please?

zhanhongliau, 2022-08-10:
This was added in `src/MarlinCore.cpp` in the setup function. Basically that setup function looks like 
```
void setup() {
  // I added these 2 lines here
  SET_INPUT_PULLDOWN(Z_MIN_PIN);
  SET_INPUT_PULLUP(Z_MIN_PIN);
  
  #ifdef BOARD_PREINIT
    BOARD_PREINIT(); // Low-level init (before serial init)
  #endif

  // ...
}
```

I haven't tested this on the latest marlin branch but it worked on [the rhapsodyv fork](https://github.com/rhapsodyv/Marlin) for my TronXY x5sa

Barrnet, 2022-08-15:
Sadly it don't work on Marlin 2.1.1 :(

closed 2023-01-03 by github-actions[bot]