MarlinFirmware/Marlin #24534: [BUG] Toggle Laser always Off and laser not turning on

GhencianTimotei, 2022-07-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When using the firmware with Laser features enable the Toggle Laser switch and the laser will stay always Off, no matter if clicked or send via terminal M3 S100.

I followed the steps described here: [https://marlinfw.org/docs/configuration/2.0.9/laser_spindle.html](https://marlinfw.org/docs/configuration/2.0.9/laser_spindle.html).

I use the pin for the z end stop. This is because I use an BLTouch and not needing an end stop button anymore + the wires are already out.
The pin used for the endstop is PA7. This is a analog compatible pin, also used for SPI communication. I tried also other pins and the behaviour is the same.

I managed to turn on the laser by sending M3 S190 or S > 190 (the power will be set depending on the power sent via terminal), but this is for only about 2 seconds. after this the laser will turn off. Meanwhile the menu will not update the setting sent over serial. Eg. the power, or the toggle state.

My printer is a Creality Ender 3 Max with the board version 4.2.2. In the past with an old version of marlin this worked without problems.

### Expected behavior

The laser should power on with the power corresponding to the "Laser Power" set from menu or via serial.

### Actual behavior

The laser will stay always off.

### Steps to Reproduce

Compile the software with the define LASER_FEATURE.
Set the define SPINDLE_LASER_USE_PWM

Set the pins:
SPINDLE_LASER_ENA_PIN PA7
SPINDLE_LASER_PWM_PIN PA7

Go to laser menu on the printer.

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Jul 21 2022 16:58:35) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 Max 4.2.2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Creality Ender 3 Max

### Electronics

Board version 4.2.2

### Add-ons

Laser added on pin PA7, BLTouch added on the stock header.

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[1623133432-Creality422-Schematic-2.pdf](https://github.com/MarlinFirmware/Marlin/files/9160101/1623133432-Creality422-Schematic-2.pdf)

[image]

[Config_files.zip](https://github.com/MarlinFirmware/Marlin/files/9160143/Config_files.zip)

cbagwell, 2022-07-21:
Newer versions have LASER_SAFETY_TIMEOUT_MS 1000 that will shut off laser very quick without movement.  That makes the menu toggle not so useful if your using it for laser focusing for example.  A value of 0 will disable it.

ellensp, 2022-07-22:
you cannot use 

SPINDLE_LASER_ENA_PIN PA7
SPINDLE_LASER_PWM_PIN PA7

the same pin cannot be used for two things

descipher, 2022-07-22:
You can define SPINDLE_LASER_ENA_PIN on an unused pin and also have a look at PR #24345

GhencianTimotei, 2022-08-04:
> Newer versions have LASER_SAFETY_TIMEOUT_MS 1000 that will shut off laser very quick without movement. That makes the menu toggle not so useful if your using it for laser focusing for example. A value of 0 will disable it.

closed 2022-08-04 by GhencianTimotei