MarlinFirmware/Marlin #24539: [BUG] G30.cpp does not compile with DWIN_CREALITY_LCD_JYERSUI enabled

TruncatedPi, 2022-07-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

 line 40 needs to be changed to 
 #if EITHER(DWIN_LCD_PROUI, DWIN_CREALITY_LCD_JYERSUI)

 so that line 94 can compile. Compile time Error is:
 Marlin\src\gcode\probe\G30.cpp:94:9: error: 'ui' was not declared in this scope

### Bug Timeline

July 22, 2022

### Expected behavior

should compile with DWIN_CREALITY_LCD_JYERSUI enabled

### Actual behavior

does not compile with DWIN_CREALITY_LCD_JYERSUI enabled

### Steps to Reproduce

1. enable DWIN_CREALITY_LCD_JYERSUI
2. compile

### Version of Marlin Firmware

latest (today)

### Printer model

Ender 3 v2

### Electronics

stock

### Add-ons

BL-Touch

### Bed Leveling

ABL Linear grid

### Your Slicer

Prusa Slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-07-24:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/24540

Bob-the-Kuhn, 2022-07-24:
Didn't fail when this config was switched to DWIN_CREALITY_LCD_JYERSUI 

https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20V2/CrealityV422/CrealityUI

thisiskeithb, 2022-07-24:
> Didn't fail when this config was switched to DWIN_CREALITY_LCD_JYERSUI

It was removed in:

- https://github.com/MarlinFirmware/Marlin/pull/24459 / https://github.com/MarlinFirmware/Marlin/commit/f73fad1b2fa6e70b3d632969ecf2773bee826643#diff-1dd233b2f2b157610bc993fedac6850a8630cfd2bf12b9b0cf9cd41353605cf6

@TruncatedPi Can you attach a config that causes this error so it can be replicated?

thisiskeithb, 2022-07-24:
@Bob-the-Kuhn: The linked config does not use a probe, but I was able to confirm the issue when one was enabled: https://github.com/MarlinFirmware/Marlin/pull/24540#issuecomment-1193353156

Closing since the PR to fix this is still needed & has been reopened.

closed 2022-07-24 by thisiskeithb