MarlinFirmware/Marlin #24543: [BUG] SKR 3 EZ + TMC2209 Sensorless homing - Sensitivity issue

jiberan1, 2022-07-24:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

After exchange motherboard from SKR 1.4 to **SKR 3 EZ sensorless homing** not work the same. Normal sensitivity for **TMC2209** is 110. After exchange the mother board it was not working - immediately triggered (X and Y). It start working from setting **sensitivity to 75 or less**. Over 75 immediately triggered. 
The issue with this low setting sensitivity is that at such a low value is **necessary higher force to trigger the endstop**. In my case the belt even starts skipping on X.

Only what has been changed is the motherboard. All hardware stay the same like before and in Marlin 2.1 (which i used and have configured before on SKR 1.4) only change the motherboard. 

### Version of Marlin Firmware

2.1

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/9176172/Marlin.zip)

thisiskeithb, 2022-07-24:
Make sure you have the diag jumpers installed & try disabling `QUICK_HOME` since that can cause issues with sensorless homing.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-07-24 by thisiskeithb

jiberan1, 2022-07-24:
Diag jumpers are installed. Sensorless homing working, but the sensitivity is issue.
Now I realize QUICK_HOME not working. Even with it switch on it still homing first X then Y. I try disabling QUICK_HOME - no change. Sensorless homing start working from sensitivity 75 or less.

EvilGremlin, 2022-07-24:
Sensetivity is different for each individual machine. But in your case something must have changed in config between boards. like _CURRENT_HOME, ot you forgot to reset EEPROM

jiberan1, 2022-07-24:
> Sensetivity is different for each individual machine. But in your case something must have changed in config between boards. like _CURRENT_HOME, or speed, or you forgot to reset EEPROM

EEPROM reset more time
_CURRENT_HOME not changed

jiberan1, 2022-07-24:
Marlin bugfix 2.1.x no change